riverloopsec/killerbee #157: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'

cn0xroot, 2019-01-11:
./zbwireshark -c 15

See the GoodFET FAQ about missing info flash.
Traceback (most recent call last):
  File "./zbwireshark", line 108, in <module>
    main()
  File "./zbwireshark", line 61, in main
    with KillerBee(device=args.devstring) as kb:
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/killerbee/__init__.py", line 167, in __init__
    raise KBInterfaceError("KillerBee doesn't know how to interact with serial device at '%s'." % self.dev)
killerbee.kbutils.KBInterfaceError: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'

system:ubuntu 12.04
devices:api-mote

cn0xroot, 2019-01-11:
sudo ./zbid 
           Dev Product String       Serial Number

See the GoodFET FAQ about missing info flash.

alphaomicron, 2019-01-19:
Is your device recognized by the host at all? have the same issue with zbid

rmspeers, 2019-01-19:
@cn0xroot  this should be working on Ubuntu 12.04 as you indicate, I suggest doing a single reflash of the device (make sure to use the appropriate goodfet.bsl options, either erase and then program and then verify, or program and verify (providing the same file as you're flashing as the password if you get that message).

rmspeers, 2019-06-02:
Is this now fixed? This looks like the OS didn't have the USB connected -- perhaps a VM without it attached or a bad cable, etc?

cn0xroot, 2019-06-13:
still unstable（not 100% work）, only  30% -50% ,Already give up to use apimote

closed 2019-06-13 by cn0xroot