riverloopsec/killerbee #159: another Api-Mote v4 Recognition Error

alphaomicron, 2019-01-18:
Got APImote from the attify store. When I connect the APImote on the host machine although the led is on (green) it's not recognized as a usb device, thus I cannot connect it to any guest OS (linux). However I did use the Attify OS and run zbid with no luck. Also tried two different cables just in case but the problem remains. Any pointers?

Thank you in advance

alphaomicron, 2019-01-19:
Drivers where installed via the ftdichip website but still no change. Device is not appearing in COM port

rmspeers, 2019-01-19:
So if you're running on Linux, you shouldn't need to install any drivers. Simply plug it in to a Ubuntu 12.04 or 16.04 (12.04 preferred) and you should see the output of the OS seeing it in dmesg. What do you see there?

alphaomicron, 2019-01-19:
0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.2.0-42-generic (buildd@lgw01-27) (gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) ) #49~14.04.1-Ubuntu SMP Wed Jun 29 20:22:18 UTC 2016 (Ubuntu 4.2.0-42.49~14.04.1-generic 4.2.8-ckt12)
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   NSC Geode by NSC
[    0.000000]   Cyrix CyrixInstead
[    0.000000]   Centaur CentaurHauls
[    0.000000]   Transmeta GenuineTMx86
[    0.000000]   Transmeta TransmetaCPU
[    0.000000]   UMC UMC UMC UMC
[    0.000000] x86/fpu: xstate_offset[2]: 0240, xstate_sizes[2]: 0100
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 0x340 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'lazy' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003ffeffff] usable
[    0.000000] BIOS-e820: [mem 0x000000003fff0000-0x000000003fffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x3fff0 max_arch_pfn = 0x1000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR variable ranges disabled:
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] MTRR: Disabled
[    0.000000] CPU MTRRs all blank - virtualized system.
[    0.000000] found SMP MP-table at [mem 0x0009fff0-0x0009ffff] mapped at [c009fff0]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x021fffff]
[    0.000000] Base memory trampoline at [c009b000] 9b000 size 16384
[    0.000000] init_memory_mapping: [mem 0x00000000-0x000fffff]
[    0.000000]  [mem 0x00000000-0x000fffff] page 4k
[    0.000000] init_memory_mapping: [mem 0x37800000-0x379fffff]
[    0.000000]  [mem 0x37800000-0x379fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x20000000-0x377fffff]
[    0.000000]  [mem 0x20000000-0x377fffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x00100000-0x1fffffff]
[    0.000000]  [mem 0x00100000-0x001fffff] page 4k
[    0.000000]  [mem 0x00200000-0x1fffffff] page 2M
[    0.000000] init_memory_mapping: [mem 0x37a00000-0x37bfdfff]
[    0.000000]  [mem 0x37a00000-0x37bfdfff] page 4k
[    0.000000] BRK [0x01ca5000, 0x01ca5fff] PGTABLE
[    0.000000] BRK [0x01ca6000, 0x01ca6fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x35972000-0x36cb0fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.000000] ACPI: XSDT 0x000000003FFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.000000] ACPI: FACP 0x000000003FFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.000000] ACPI: DSDT 0x000000003FFF0470 0021FF (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.000000] ACPI: FACS 0x000000003FFF0200 000040
[    0.000000] ACPI: FACS 0x000000003FFF0200 000040
[    0.000000] ACPI: APIC 0x000000003FFF0240 000054 (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.000000] ACPI: SSDT 0x000000003FFF02A0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] 131MB HIGHMEM available.
[    0.000000] 891MB LOWMEM available.
[... 461 lines cut ...]
[   18.944323] snd_intel8x0 0000:00:05.0: white list rate for 1028:0177 is 48000
[   19.328786] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   19.329664] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   19.330021] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   19.741339] floppy0: no floppy controllers found
[   19.741376] work still pending
[   21.164427] vboxsf: Successfully loaded version 5.2.22 (interface 0x00010004)
[   21.499156] init: plymouth-upstart-bridge main process (153) killed by TERM signal
[   33.432338] VBoxService 5.2.22 r126460 (verbosity: 0) linux.x86 (Nov  8 2018 20:51:07) release log
[   33.432338] 00:00:00.000082 main     Log opened 2019-01-19T16:16:53.921459000Z
[   33.432383] 00:00:00.000158 main     OS Product: Linux
[   33.432406] 00:00:00.000186 main     OS Release: 4.2.0-42-generic
[   33.432429] 00:00:00.000208 main     OS Version: #49~14.04.1-Ubuntu SMP Wed Jun 29 20:22:18 UTC 2016
[... cut ...]
[   48.073961] audit: type=1400 audit(1547914628.558:64): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=2573 comm="apparmor_parser"
[   48.073978] audit: type=1400 audit(1547914628.558:65): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2573 comm="apparmor_parser"
[   48.081083] audit: type=1400 audit(1547914628.566:66): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2573 comm="apparmor_parser"
[   53.884295] ISO 9660 Extensions: Microsoft Joliet Level 3
[   53.932472] ISO 9660 Extensions: RRIP_1991A

alphaomicron, 2019-01-20:
So managed to set up a pure Ubuntu 12.04 box. No luck there either
APImote
Ubuntu 12.04

dmesg

[    0.731182] microcode: Microcode Update Driver: v2.00 <tigran@aivazian.fsnet.co.uk>, Peter Oruba
[    0.731184] Scanning for low memory corruption every 60 seconds
[    0.731484] Initialise system trusted keyring
[    0.731529] audit: initializing netlink socket (disabled)
[    0.731537] type=2000 audit(1547994015.588:1): initialized
[    0.756418] HugeTLB registered 2 MB page size, pre-allocated 0 pages
[    0.757551] zbud: loaded
[    0.757671] VFS: Disk quotas dquot_6.5.2
[    0.757710] Dquot-cache hash table entries: 512 (order 0, 4096 bytes)
[    0.758081] fuse init (API version 7.22)
[    0.758144] msgmni has been set to 24009
[    0.758189] Key type big_key registered
[    0.758521] Key type asymmetric registered
[    0.758522] Asymmetric key parser 'x509' registered
[    0.758548] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
[    0.758589] io scheduler noop registered
[    0.758590] io scheduler deadline registered (default)
[    0.758611] io scheduler cfq registered
[    0.758782] pcieport 0000:00:1c.0: enabling device (0106 -> 0107)
[    0.758925] pcieport 0000:00:1c.0: irq 64 for MSI/MSI-X
[    0.759073] pcieport 0000:00:1c.2: irq 65 for MSI/MSI-X
[    0.759233] pcieport 0000:00:1c.3: irq 66 for MSI/MSI-X
[    0.759379] pcieport 0000:00:1c.4: irq 67 for MSI/MSI-X
[    0.759440] pci_hotplug: PCI Hot Plug PCI Core version: 0.5
[    0.759451] pciehp: PCI Express Hot Plug Controller Driver version: 0.4
[    0.759485] vesafb: mode is 640x480x32, linelength=2560, pages=0
[    0.759487] vesafb: scrolling: redraw
[    0.759488] vesafb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    0.759559] vesafb: framebuffer at 0xf9000000, mapped to 0xffffc90011c00000, using 1216k, total 1216k
[    0.805703] Console: switching to colour frame buffer device 80x30
[    0.851836] fb0: VESA VGA frame buffer device
[    0.851849] intel_idle: MWAIT substates: 0x1120
[    0.851860] intel_idle: v0.4 model 0x1A
[    0.851861] intel_idle: lapic_timer_reliable_states 0x2
[    0.852023] ipmi message handler version 39.2
[    0.852081] input: Power Button as /devices/LNXSYSTM:00/device:00/PNP0C0C:00/input/input0
[    0.852085] ACPI: Power Button [PWRB]
[    0.852111] input: Power Button as /devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
[    0.852113] ACPI: Power Button [PWRF]
[    0.852261] GHES: HEST is not enabled!
[    0.852322] Serial: 8250/16550 driver, 32 ports, IRQ sharing enabled
[    0.853503] Linux agpgart interface v0.103
[    0.854666] brd: module loaded
[    0.855296] loop: module loaded
[    0.855398] ata_piix 0000:00:1f.2: version 2.13
[    0.855501] ata_piix 0000:00:1f.2: MAP [ P0 P2 P1 P3 ]
[    0.856223] scsi0 : ata_piix
[    0.856281] scsi1 : ata_piix
[    0.856306] ata1: SATA max UDMA/133 cmd 0x9000 ctl 0x8c00 bmdma 0x8480 irq 20
[    0.856310] ata2: SATA max UDMA/133 cmd 0x8880 ctl 0x8800 bmdma 0x8488 irq 20
[    0.856411] ata_piix 0000:00:1f.5: MAP [ P0 -- P1 -- ]
[    0.856967] scsi2 : ata_piix
[    0.857092] scsi3 : ata_piix
[    0.857116] ata3: SATA max UDMA/133 cmd 0xa000 ctl 0x9c00 bmdma 0x9480 irq 20
[    0.857119] ata4: SATA max UDMA/133 cmd 0x9880 ctl 0x9800 bmdma 0x9488 irq 20
[    0.857330] libphy: Fixed MDIO Bus: probed
[    0.857393] tun: Universal TUN/TAP device driver, 1.6
[    0.857394] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[    0.857431] PPP generic driver version 2.4.2
[    0.857466] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.857470] ehci-pci: EHCI PCI platform driver
[    0.857589] ehci-pci 0000:00:1a.7: EHCI Host Controller
[    0.857595] ehci-pci 0000:00:1a.7: new USB bus registered, assigned bus number 1
[    0.857606] ehci-pci 0000:00:1a.7: debug port 1
[... 305 lines cut ...]
[    3.985313] nouveau  [     DRM] TMDS table version 2.0
[    3.985315] nouveau  [     DRM] DCB version 4.0
[    3.985318] nouveau  [     DRM] DCB outp 00: 02000300 00000028
[    3.985321] nouveau  [     DRM] DCB outp 01: 01000302 00000030
[    3.985323] nouveau  [     DRM] DCB outp 02: 04011310 00000028
[    3.985326] nouveau  [     DRM] DCB conn 00: 00001030
[    3.985329] nouveau  [     DRM] DCB conn 01: 00000100
[    4.025709] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    4.025710] [drm] No driver support for vblank timestamp query.
[    4.082474] nouveau  [     DRM] MM: using M2MF for buffer copies
[    4.123282] nouveau  [     DRM] allocated 1024x768 fb: 0x70000, bo ffff88032be1a000
[    4.123478] fbcon: nouveaufb (fb0) is primary device
[    4.193602] Console: switching to colour frame buffer device 128x48
[    4.193883] nouveau 0000:02:00.0: fb0: nouveaufb frame buffer device
[    4.193885] nouveau 0000:02:00.0: registered panic notifier
[    4.193891] [drm] Initialized nouveau 1.1.1 20120801 for 0000:02:00.0 on minor 0
[    6.400093] wlan0: authenticate with ----------
[    6.442499] wlan0: direct probe to ----------- (try 1/3)
[    6.642977] wlan0: send auth to------------(try 2/3)
[    6.646501] wlan0: authenticated
[    6.647004] wlan0: associate with----------- (try 1/3)
[    6.656101] wlan0: RX AssocResp from ---------- (capab=0x411 status=0 aid=6)
[    6.663344] 
[    6.663388] IPv6: ADDRCONF(NETDEV_CHANGE): 
[    7.501482] input: PS/2 Generic Mouse as /devices/platform/i8042/serio1/input/input4

alphaomicron, 2019-01-20:
Ubuntu 12.04
APImote v4

dmesg

[    0.853503] Linux agpgart interface v0.103
[    0.854666] brd: module loaded
[    0.855296] loop: module loaded
[    0.855398] ata_piix 0000:00:1f.2: version 2.13
[    0.855501] ata_piix 0000:00:1f.2: MAP [ P0 P2 P1 P3 ]
[    0.856223] scsi0 : ata_piix
[    0.856281] scsi1 : ata_piix
[    0.856411] ata_piix 0000:00:1f.5: MAP [ P0 -- P1 -- ]
[    0.856967] scsi2 : ata_piix
[    0.857092] scsi3 : ata_piix
[    0.857116] ata3: SATA max UDMA/133 cmd 0xa000 ctl 0x9c00 bmdma 0x9480 irq 20
[    0.857119] ata4: SATA max UDMA/133 cmd 0x9880 ctl 0x9800 bmdma 0x9488 irq 20
[    0.857330] libphy: Fixed MDIO Bus: probed
[    0.857393] tun: Universal TUN/TAP device driver, 1.6
[    0.857394] tun: (C) 1999-2004
[    0.857431] PPP generic driver version 2.4.2
[    0.857466] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.857470] ehci-pci: EHCI PCI platform driver
[    0.857589] ehci-pci 0000:00:1a.7: EHCI Host Controller
[    0.857595] ehci-pci 0000:00:1a.7: new USB bus registered, assigned bus number 1
[    0.857606] ehci-pci 0000:00:1a.7: debug port 1
[    0.861495] ehci-pci 0000:00:1a.7: cache line size of 64 is not supported
[    0.861510] ehci-pci 0000:00:1a.7: irq 18, io mem 0xf7fff000
[    0.871021] ehci-pci 0000:00:1a.7: USB 2.0 started, EHCI 1.00
[    0.871065] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    0.871067] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.871069] usb usb1: Product: EHCI Host Controller
[    0.871070] usb usb1: Manufacturer: Linux 3.13.0-32-generic ehci_hcd
[    0.871072] usb usb1: SerialNumber: 0000:00:1a.7
[    0.871217] hub 1-0:1.0: USB hub found
[    0.871225] hub 1-0:1.0: 6 ports detected
[    0.871437] ehci-pci 0000:00:1d.7: EHCI Host Controller
[    0.871442] ehci-pci 0000:00:1d.7: new USB bus registered, assigned bus number 2
[    0.871453] ehci-pci 0000:00:1d.7: debug port 1
[    0.875340] ehci-pci 0000:00:1d.7: cache line size of 64 is not supported
[    0.875352] ehci-pci 0000:00:1d.7: irq 23, io mem 0xf7ffe000
[    0.887077] ehci-pci 0000:00:1d.7: USB 2.0 started, EHCI 1.00
[    0.887130] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002
[    0.887132] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.887134] usb usb2: Product: EHCI Host Controller
[    0.887136] usb usb2: Manufacturer: Linux 3.13.0-32-generic ehci_hcd
[    0.887137] usb usb2: SerialNumber:
[    0.887360] hub 2-0:1.0: USB hub found
[    0.887368] hub 2-0:1.0: 6 ports detected
[    0.887471] ehci-platform: EHCI generic platform driver
[    0.887477] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.887478] ohci-pci: OHCI PCI platform driver
[    0.887486] ohci-platform: OHCI generic platform driver
[    0.887490] uhci_hcd: USB Universal Host Controller Interface driver
[    0.887596] uhci_hcd 0000:00:1a.0: UHCI Host Controller
[    0.887601] uhci_hcd 0000:00:1a.0: new USB bus registered, assigned bus number 3
[    0.887630] uhci_hcd 0000:00:1a.0: irq 16, io base 0x0000a800
[    0.887668] usb usb3: New USB device found, idVendor=1d6b, idProduct=0001
[    0.887670] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.887671] usb usb3: Product: UHCI Host Controller
[    0.887673] usb usb3: Manufacturer: Linux 3.13.0-32-generic uhci_hcd
[    0.887674] usb usb3: SerialNumber: 
[    0.887876] hub 3-0:1.0: USB hub found
[    0.887886] hub 3-0:1.0: 2 ports detected
[    0.888048] uhci_hcd 0000:00:1a.1: UHCI Host Controller
[    0.888053] uhci_hcd 0000:00:1a.1: new USB bus registered, assigned bus number 4
[    0.888081] uhci_hcd 0000:00:1a.1: irq 21, io base 0x0000a880
[    0.888119] usb usb4: New USB device found, idVendor=1d6b, idProduct=0001
[    0.888121] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.888123] usb usb4: Product: UHCI Host Controller
[... 251 lines cut ...]
[    3.984909] nouveau  [  PTHERM][0000:02:00.0] internal sensor
[    3.984917] nouveau  [     CLK][0000:02:00.0] 0f: core 567 MHz shader 1350 MHz memory 400 MHz
[    3.984996] nouveau  [     CLK][0000:02:00.0] --: core 566 MHz shader 1350 MHz memory 399 MHz
[    3.985283] [TTM] Zone  kernel: Available graphics memory:
[    3.985286] [TTM] Zone   dma32: Available graphics memory: 
[    3.985287] [TTM] Initializing pool allocator
[    3.985293] [TTM] Initializing DMA pool allocator
[    3.985305] nouveau  [     DRM] VRAM: 512 MiB
[    3.985308] nouveau  [     DRM] GART: 1048576 MiB
[    3.985313] nouveau  [     DRM] TMDS table version 2.0
[    3.985315] nouveau  [     DRM] DCB version 4.0
[    3.985318] nouveau  [     DRM] DCB outp 00: 02000300 00000028
[    3.985321] nouveau  [     DRM] DCB outp 01: 01000302 00000030
[    3.985323] nouveau  [     DRM] DCB outp 02: 04011310 00000028
[    3.985326] nouveau  [     DRM] DCB conn 00: 00001030
[    3.985329] nouveau  [     DRM] DCB conn 01: 00000100
[    4.025709] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    4.025710] [drm] No driver support for vblank timestamp query.
[    4.082474] nouveau  [     DRM] MM: using M2MF for buffer copies
[    4.123282] nouveau  [     DRM] allocated 1024x768 fb: 0x70000, bo ffff88032be1a000
[    4.123478] fbcon: nouveaufb (fb0) is primary device
[    4.193602] Console: switching to colour frame buffer device 128x48
[    4.193883] nouveau 0000:02:00.0: fb0: nouveaufb frame buffer device
[    4.193885] nouveau 0000:02:00.0: registered panic notifier
[    4.193891] [drm] Initialized nouveau 1.1.1 20120801 for 0000:02:00.0 on minor 0

cldrn, 2019-02-17:
What did you learn from this report? I'm in exactly the same situation. Not recognized, only green light.

rmspeers, 2019-02-17:
Try doing a full erase and reflash of the device with the firmware in the repository (`-e -p ...`) and share output here. Also try running a `board=apimote2 ./goodfet.ccspi info` command using the code from the GoodFET repository's client folder and share that output.

alphaomicron, 2019-02-17:
Device was replaced by a new one. As far as what caused this. I am not sure what the findings where...There was a thin protection wrap when it was delivered, which was insufficient. This could have caused some damage to the original board. Ryan was kind enough to verify with the  store and arrange a replacement.

cldrn, 2019-02-23:
> Try doing a full erase and reflash of the device with the firmware in the repository (`-e -p ...`) and share output here. Also try running a `board=apimote2 ./goodfet.ccspi info` command using the code from the GoodFET repository's client folder and share that output.

Nothing gets detected when I connect it to the host (Windows 10/OS X host). What would you try next? I believe the card might be damaged as well.

taylorcenters, 2021-07-01:
Original issue has been resolved. If you are still seeing an issue on the latest software please open a new issue.

closed 2021-07-01 by taylorcenters