riverloopsec/killerbee #160: Trying to get Api-Mote v4 Beta running on xubuntu 18.04

favid1992, 2019-01-24:
Hi,

I am trying to get the Api-Mote running on an Xubuntu 18.04. 

First of all, I had trouble to get the firmware running on the API-Mote, but with help from the issue 145, I could get it working. #145 

Unfortunately when I try to run, for example, zbstumbler, I get an interface Error:
"See the GoodFET FAQ about missing info flash.
Interface Error: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'."

Could someone help me to get it running? Or is it even possible on this OS?!

Cheers,

David

rmspeers, 2019-01-24:
It sounds like you can communicate to the device then, although I'm wondering why you needed to reflash -- that is often what gets the info flash issue happening.

Although we expect issues with the Ubuntu 18.04 kernel, there is nothing that should prevent it from working at all, but you likely will need to experiment and update some of the below:
- the FTDI chip's configuration
- the pySerial/pyUSB version or commands used to enumerate/interact with the device
- possibly udev rules to ensure there are no contention issues

favid1992, 2019-01-24:
I reflashed it because the API-Mote could not communicate with my OS at all...

Thanks for your quick response. First I need to understand the things you just named and then I am going to experiment with it. 

I will keep you updated.

favid1992, 2019-02-14:
Hey @rfmcpherson ,

I got the Api-Mote working. But now I have got the following error:

Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
issue in peeking for a register
list index out of range
ERROR: Unable to inject packet: list index out of range

Do you know this error and what I could do to get rid of it? 

Cheers,

David

rmspeers, 2019-02-14:
I'm interested to hear how you got the ApiMote working on your system -- can you update that to help others with that process on your OS?

Those errors are coming from the GoodFET Python code that we use to communicate to the ApiMote. It looks to me that it's not actually communicating with the device. Try running `board=apimote2 ./goodfet.ccspi rssi` or something to ensure communication, at least via that path, is working. I'm wondering if there are any code differences between that and how KillerBee invokes the GoodFETCCSPI.py files if so.

favid1992, 2019-02-15:
I gave up trying on the 18.04 version, so I switched to 16.04. Flashed the Api-Mote again and now it looks like it is working sometimes and sometimes not. (See attached file)
[image]

Could you explain this behavior? Why it is not able to communicate with the GoodFET code and then it runs without an error and last but not least there was no communication with the board at all? I got those messages within 5 minutes of testing. 

I was also not able to execute your named command. Is there another way to test the communication?

Out of curiosity, I tried out other tools from the framework and "zbwireshark" was the only tool which runs without returning an error.

Cheers,

rmspeers, 2019-02-15:
As best as we know, the behavior has to do with the FTDI chip and how it synchronizes with your computer. When it gets into a good state, it's reliable -- but the issue is on getting the initial communication up. This is why you sometimes see it connect and sometimes not. Specifying a -i option to tell it the interface, e.g. `-i /dev/ttyUSB0` may help as it has to less auto-discovery in that case.

For the command I listed, clone the goodfet code repo and access that command inside the client folder.

fsevb, 2019-03-14:
Hey,
I took over trying to debug the device in question. I tried everything you explained and did some own tests. Here is what I found out yet:

- FTDI Chip and drivers seems to work. Tried on Windows 10, Xubuntu 18.04/16.04 aswell as a Kali VM. No problems recognizing the device and choosing the right driver.

- GoodFET seems to work aswell. Tried running goodfet.ccspi with different params and never triggered an error. The returned values were always right.

- Killerbee Framework: 
Here's the point of trouble. The function pyserInit() in GoodFET.py can not reliably make a connection to the Apimote. I found no way to reproduce the error it seems rather random. Whenever I use a Killerbee tool it seems to fail at the point of establishing the serial connection.

The only reproducible error I found was when calling the show_dev() function of the Killerbee class. It only worked every second try, otherwise giving the "See the GoodFET FAQ..." error.

At this point I have no idea where to further look for the error, as the goodfet library seems to work, the board seems to work, but killerbee is not able to connect to the board.

Kind regards,
Felix

rmspeers, 2019-03-14:
Perhaps the GoodFET repo and KillerBee repo are using slightly different GoodFET.py files or something? Both goodfet.ccspi and the KillerBee code should be calling into the same GoodFET.py code (KillerBee repo just has a copy of it, but it’s from a while ago).

The other idea is that auto-discovery which KillerBee has to do is messing things up. You could hardcode the endpoint with -i flags (like GoodFET does via the GOODFET environment variable I think), and then tell KillerBee to not try to figure out what type the serial device is by changing the initialization code.

Thanks for narrowing this down — hopefully this item is something we could fix with some help from your debugging.

fsevb, 2019-03-15:
Via md5sum I verified that the GoodFET.py files are indeed different. However I failed to locate the actual file used by Killerbee.

Furthermore I discovered, that the Apimote is missing the 8-pin chip between the yC and the FTDI-Chip. I assume there should be a SST25 Flash chip. Could this be the problem?

amitSmartthings, 2019-04-11:
I am facing the exact same issue.
The response for zstumbler varies from time to time.
At times I get:
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
issue in peeking for a register
list index out of range
ERROR: Unable to inject packet: list index out of range

OR

See the GoodFET FAQ about missing info flash.
Interface Error: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'.

OR

Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbstumbler')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 140, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 156, in __init__
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 355, in isgoodfetccspi
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 134, in serInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 216, in pyserInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 408, in telosBReset
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 308, in telosI2CWriteCmd
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 297, in telosI2CWriteByte
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 290, in telosI2CWriteBit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 271, in telosSetSCL
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 596, in setRTS
    self.rts = value
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 459, in rts
    self._update_rts_state()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 629, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_RTS_str)
IOError: [Errno 5] Input/output error

Was anyone able to find a solution for this?

amit821990, 2019-04-11:
I am facing a similar issue to what to described above.
Responses for zbstumbler vary from time to time. Here I the responses I received:

See the GoodFET FAQ about missing info flash.
Interface Error: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'.

OR

zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
issue in peeking for a register
list index out of range
ERROR: Unable to inject packet: list index out of range

OR
Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.0', 'zbstumbler')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.0-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 140, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 156, in __init__
  File "build/bdist.linux-x86_64/egg/killerbee/kbutils.py", line 355, in isgoodfetccspi
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 134, in serInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 216, in pyserInit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 407, in telosBReset
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 308, in telosI2CWriteCmd
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 298, in telosI2CWriteByte
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 287, in telosI2CWriteBit
  File "build/bdist.linux-x86_64/egg/killerbee/GoodFET.py", line 271, in telosSetSCL
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 596, in setRTS
    self.rts = value
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 459, in rts
    self._update_rts_state()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 627, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
IOError: [Errno 5] Input/output error

I am using Ubuntu 18.04.2.
Was anybody able to figure out a solution for this?

riverloopsec, 2019-04-12:
> Furthermore I discovered, that the Apimote is missing the 8-pin chip between the yC and the FTDI-Chip. I assume there should be a SST25 Flash chip. Could this be the problem?

That SST25 is not an issue, it is an optional component that is not populated unless needed (to reduce board cost). It is something that can be populated and then used for logging/storage of packets if the device is being run off of battery.

riverloopsec, 2019-04-12:
> I am using Ubuntu 18.04.2.

Can you try using the Ubuntu 12.04 that we ask people to test on as a base? Are you on a VM?

Promobyte, 2019-04-19:
Replacing all GoodFET .py files in /killerbee with the files from travisgoodspeed/goodfet.git worked for me. 

Running Ubuntu 18.04.2 TLS

zbid, zbstumbler, zbwireshark working consistently.

taylorcenters, 2021-07-01:
We believe that the most recent PRs have addressed this apimote instability problem. Please try the latest software and reopen if this is still an issue.

closed 2021-07-01 by taylorcenters