riverloopsec/killerbee #161: Add GoodFET firmware source code and configuration

yannayl, 2019-01-28:
The GoodFET firmware has some bugs, e.g. see https://github.com/travisgoodspeed/goodfet/issues/42
In order to fix and test the fix of such issues, we need to be able to build the firmware ourselves.

Currently, the source code of the GoodFET and the configuration for the various supported boards is not included in this repository. Furthermore, the GoodFET version used is to build the firmware images is not specified, only the client.

Please add these to the repository.

rmspeers, 2019-01-28:
The GoodFET firmware source code, as described in the KillerBee project, is maintained in the GoodFET repository which you linked. That has all of the source code needed to build it. It also has the configuration needed, see https://github.com/travisgoodspeed/goodfet/blob/master/firmware/config.mk#L16. Setting those board variables during build is all you need to do AFAIK.

yannayl, 2019-01-28:
Thanks!

I was somewhat baffled because you do keep the client code and the binary image in this repo.

closed 2019-01-28 by yannayl