riverloopsec/killerbee #162: RZUSB/CC2531 not receiving frames between 55-64 bytes

hgbstu, 2019-02-01:
#### Brief description

Killerbee doesn't receive frames with length between 55-64 bytes.

#### Environment

-   Python version: Python 2.7.13 
-   Operating System: Debian 9.5
-   Sniffing devices: CC2531EMK, RZRAVENUSB Killerbee006

#### Description
While injecting ZigBee frames I noticed that I don't see the injected frames in in my pcap. I was sniffing with one RZRAVENUSB and injecting frames with another one. So I tried injecting frames with different lengths and noticed that frames with 54 bytes and below are sniffed correctly, frames with a length of 55, 56, 58, 60, 62, 64 bytes are missing in my pcap. I didn't try sending frames with 57, 59, 61, 63 bytes.

I validated that the frames are injected correctly with Ubiqua and a CC2531EMK stick, I can see the frames in Ubiqua.
Therefore I thought that it is a RZRAVENUSB specific issue, but it's also the same when I try to sniff traffic with zbdump and a CC2531 stick. 

I also checked my previous ZigBee pcaps and couldn't find a single frame with a size between 55 and 64 bytes. I know for sure that there were some frames transmitted with a length between 55 and 64 bytes because I was sniffing with an SDR simultaneously.

Could you guys check if you can reproduce this behaviour? I would gladly investigate the issue further, just tell me how :) 

#### Incorrect result
```
zbdump -i "2:25" -w /home/user/killerbee/captures/cc2531_channel_25.pcap -c 25 -v
Warning: You are using pyUSB 1.x, support is in beta.
zbdump: listening on 'CC2531 USB Dongle', channel 25, page 0 (2475.0 MHz), link-type DLT_IEEE802_15_4, capture size 127 bytes
.....................................................................................................................................^C
```
[cc2531_zbdump_channel_25.zip](https://github.com/riverloopsec/killerbee/files/2822542/cc2531_zbdump_channel_25.zip)

#### Expected result
[cc2531_ubiqua_channel_25.zip](https://github.com/riverloopsec/killerbee/files/2822541/cc2531_ubiqua_channel_25.zip)

hgbstu, 2019-05-24:
I investigated this issue a little further and found the reason for this bug.

killerbee/dev_rzusbstick.py:
```
#TODO added len(pdata) check as some arrays were failing
if pdata == None or pdata == () or len(pdata) == 0 or len(pdata) <= 10:
```

The len(pdata) check fails on frames with that exact length. I uncommented this check and now I can see the missing frames.

@rmspeers @Crypt0s Do you know why this check has been put in place?

rmspeers, 2019-05-26:
What frames did you want to capture of that length?
I don’t recall why that specific length... would have thought that if so it
would have been some minimum frame length from the chip which is why I’m
interested in the short frames you want to see - I assume things like ACKs?

hgbstu, 2019-05-27:
The code I mentioned affects specific frame lengths (I tested frames with a frame length of 55-64bytes)  which are too big for a single USB Request Block. Therefore, the incoming frame is split into multiple USB Request Blocks.

For example a Device Announcement frame with 57 bytes is split into two USB Request Blocks. The first one contains the first bytes and the second USB Request Block only contains 3 bytes (2 bytes FCS and  0xFF that is discarded). As the payload of the second USB Request Block is smaller than 10 bytes it is discarded by Killerbee and the frame is never returned in pnext().

See attached capture that contains the device announcement (captured via cc2531 and ubiqua) and the corresponding usbmon log when rzusbstick is used in killerbee
[example_device_announcement.zip](https://github.com/riverloopsec/killerbee/files/3221364/example_device_announcement.zip)

taylorcenters, 2021-06-22:
Did we figure out why the len check was necessary in the original code, and if we want to remove it to support cases like this where real data is being lost?

taylorcenters, 2022-03-08:
removed dropping small packets in https://github.com/riverloopsec/killerbee/tree/bugfix/cc2531-assorted

closed 2022-03-08 by taylorcenters