riverloopsec/killerbee #163: zbid output on APImote v4

alphaomicron, 2019-02-17:
OS: debian, ubuntu
Device: APImote v4 beta

While giving command zbid The output is
`zbid
           Dev Product String       Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2            
`
Is this correct? I thought it should give v4

fhred, 2019-02-20:
I'm not 100% sure, but I think the `v2` there relates to the firmware version. Read it like: 

GoodFET for Api-Mote version 2

rmspeers, 2019-02-20:
Output of v2 for zbid is expected, although perhaps a better output would be "v2+". Basically v1 and v2-v4 have to be enumerated differently and communicated with differently, so that's all it's saying. v2, 3, and 4 are all communicated with the same way.

alphaomicron, 2019-02-20:
So does this mean I have to flash a newer firmware?

rmspeers, 2019-02-20:
No, these devices are shipped with up-to-date firmware and only should need reflashing if you have updated the firmware and want to add features/etc.

rmspeers, 2019-06-02:
Closing. New FAQ will also remind people of this.

closed 2019-06-02 by rmspeers