riverloopsec/killerbee #166: Problem after flashing

bigthreede, 2019-03-18:
I had a green LED (D103) lighting on a APImote v4 beta, but device was not detected by zbid. So i flashed apimotev4_gf.hex to the board, but now green LED is off, and LED D303 is red. How can i get the board to run?

masticover, 2019-04-04:
I've purchased a new one and I only have the RED LED shining... is it the expected behavior?
Thanks.

bigthreede, 2019-04-04:
As far as i know only the green D103 should be on. Can you rech your device with zbid command? (included in killerbee)

masticover, 2019-04-04:
This is the output for zbid command:
Dev Product String       Serial Number
/dev/ttyUSB0 GoodFET Api-Mote v2

And here is the output for zbstumbler command:
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
issue in peeking for a register
list index out of range
ERROR: Unable to inject packet: list index out of range

This is a just purchased new brand Apimote. Isn't it supposed to have v4? Anything else to know about how to make it work?

Thanks a lot for your help.

bigthreede, 2019-04-05:
It looks your device is working, even without the green LED, zbid detects it. Don't worry if it's shown as an apimote v2, be sure it is an v4, it's just shown wrong. zbstumbler needs as far as  i know some input parameters like channel etc., but i don't know in detail, i did not come that far. You need to read how to use the zigbee tools, i'm sorry i can't help you here.

masticover, 2019-04-08:
Ok, thanks anyway for the information. I will try to experiment with the device and see if more issues arise.

rmspeers, 2019-06-02:
Is this now working for you?  The device, as @bigthreede  says, is working as shown in zbid. The zbstumbler run didn't appear to get a good serial sync (aka the exact reason it is a "beta" in Apimote v4beta), but using `-i` with the command when doing zbstumbler is best practice, but still may need to reconnect a few times to get it synced up.

inode-, 2020-11-18:
Workaround at: https://github.com/riverloopsec/killerbee/issues/215#issuecomment-729582886

taylorcenters, 2021-06-22:
closed - workaround in #215 will continue working on it there

closed 2021-06-22 by taylorcenters