riverloopsec/killerbee #169: zbstumbler not finding anything, just iterating transmission on different channels

nasharam, 2019-05-19:
hi,
i was just getting started with the apimote v4 beta board but when i ran zbstumbler -v it is repeatedly transmitting the beacon request but not finding anything. no matter what how long i keep it running, no changes in result.
i'm using nano io shield with xbee and arduino nano to send analogueserial message which is working fine which i can see in serial monitor of arduino ide. 
hope someone here has some solution for my problem
 (this is done in ubuntu vm)

rmspeers, 2019-06-02:
This report appears like an introductory question similar to https://github.com/riverloopsec/killerbee/issues/171. Please read the response there. Also note that xBee, depending on version, isn't going to implement standard ZigBee. Thus you may need to alter the tools to detect it.

r00tb3, 2019-08-24:
> Thus you may need to alter the tools to detect it.

When you say tools, do you mean the XBee modules configuration OR ApiMote?

If it's ApiMote then what changes do we need to make in it to match the same config as XBee module's Zigbee config?

Regards.

taylorcenters, 2021-07-01:
Hi, we believe that the apimote instability issues have been fixed with the latest PRs. Please try the latest software and reopen the issue if this is still a problem. Thanks

closed 2021-07-01 by taylorcenters