riverloopsec/killerbee #171: zstumbler output

amam76, 2019-06-02:
Hi,

I bought APIMote and I am trying to sniff Zigbee traffic from Philips Hue bulb G1. When I use zbstumbler, I get the following output. There is something on channel 15 but it seems that it has no data.
I am not using an antenna. 

1)	Can anyone suggest what could be the problem?
2)	I want to try using a antenna but I am not sure which one to use.
a.	Can anyone guide me to an antenna that is supported by APIMote that I can find online?
b.	I understand that we can use the internal antenna as an alternative. Can anyone help me how to make sure that internal antenna is selected?

I am running an up to date Kali Linux on Vmware VM and host OS is Windows 7.

Thank you for your support!

root@kali:~/killerbee# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 007: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 005: ID 0e0f:0008 VMware, Inc. 
Bus 001 Device 004: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

root@kali:~# zbstumbler -v
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
Setting channel to 12.
Transmitting beacon request.
Setting channel to 13.
Transmitting beacon request.
Setting channel to 14.
Transmitting beacon request.
Setting channel to 15.
Transmitting beacon request.
# DEBUG Clearing overflow
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.

root@kali:~# zbstumbler -c 15 -v
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Transmitting beacon request.
# DEBUG Clearing overflow
Transmitting beacon request.
Transmitting beacon request.
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.
Received frame is not a beacon (FCF=4188).
Transmitting beacon request.
# DEBUG Clearing overflow
Transmitting beacon request.

rmspeers, 2019-06-02:
As you are trying to capture RF, we strongly suggest using an antenna -- either moving the ApiMote's component that selects the internal antenna, or by adding an external RP-SMA antenna (which is what is expected by default). Not doing one of those will result in poor performance, missed packets, potential corrupted packets being received, etc.

# Internal antenna:
To select the internal antenna, you would move C501 to the other position, as shown in the RF Zone page of https://github.com/riverloopsec/apimote/blob/master/hardware/apimotev4beta/run4.pdf.

# External antenna:
Any RP-SMA 2.4 GHz antenna will work. A Google search shows these available on Adafruit, DigiKey, eBay, Amazon, and more.

rmspeers, 2019-06-02:
Also, with regards to why you are not seeing beacon responses in zbstumbler, I suggest you analyze the traffic at the packet level (e.g., zbdump, etc) and look at if it is sending standard beacon frames to you or not.

closed 2019-06-02 by rmspeers