riverloopsec/killerbee #172: Cannot extract network key

Bking82, 2019-06-12:
Hi
I'm using the API-Mote to capture traffic from my Philips Hue hub to a Hue bulb. When viewing this in Wireshark, i don't see the network key exchange. 
I've tried deleting the bulb from the Hue app so the bulb will have to rejoin the network (and thus receive the network key encrypted with the ZLL key), and also resetting it using the API Debug tool as described here:
[https://community.smartthings.com/t/hue-bulb-factory-reset-solved-if-your-hue-bulb-is-on-a-zll-channel/84967](url)
ZBsniff also just returns that the 1 PCAP file was processed, with no more info.
I have added three keys into Wireshark's Zigbee pre-configured keys as described here:
[https://hal9k.dk/sniffing-philips-hue-zigbee-traffic-with-wireshark/](url)
I'm really stumped. Any help would be greatly appreciated.
I'm quite new to Linux, and am running AttifyOS via VirtualBox (if that helps).
Brian

Bking82, 2019-06-13:
Just to add, i'm getting a lot of "Bad FCS" on IEEE packets, similar to those reported in issue #139. Was this ever resolved? I'm assuming some of these packets could be the network key exchange?

taylorcenters, 2021-07-01:
Hi, we believe that the apimote instability issues have been fixed with the latest PRs. Please try the latest software and reopen the issue if this is still a problem. Thanks

closed 2021-07-01 by taylorcenters