riverloopsec/killerbee #173: Api-Mote cannot correctly sniff packets

UgurCIL, 2019-06-28:
Hello,
I am a newbie for ZigBee, Api-Mote, CC2531 etc. But I think I have a problem with my Api-Mote sniffer. It cannot correctly capture ZigBee packets.
I have been trying to sniff traffic of Xiaomi Smart Home Kit. Two different sniffer (CC2531 and Api-Mote) give different packet list.

sudo zbwireshark -c 20 -i 2:6 [for CC2531]
sudo zbwireshark -c 20 -i /dev/ttyUSB [for Api-Mote]

I have run those commands simultaneously and added the captured packet below. Packet 3 in apimote.pcap is the concatenation of packet 5, '\x50', and packet 6 in cc2531.pcap
This is also valid for packet 4 in apimote and packet 7-8 in cc2531.pcap

I think Api-Mote represents the ZCL: Report and ACK in one packet. 

[packets.zip](https://github.com/riverloopsec/killerbee/files/3339347/packets.zip)
[image]

riverloopsec, 2019-06-28:
What is your setup on the ApiMote? It looks like it is getting corrupted
packets over the air - due to bad reception probably.
It could also be that it’s having trouble keeping up on the packet rate.
You’ll notice it looks like it’s missing some packets and getting
corruptions on others.

UgurCIL, 2019-06-28:
Thanks for really quick response,

But I cannot understand what you mean by typing "setup on ApiMote". I just plugged in and used it. My host is ubuntu 18.04 also I have the same result when using AttifyOS or ubuntu 16.04
If we are talking about pins and switches on ApiMote, picture is below. 

packet rate is very low. for example, cc2531 can handle this environment.
[image]

rmspeers, 2019-07-01:
Ok, at least it looks like you have an antenna connected, which I presume is correctly an RP-SMA and 2.4 GHz tuned.

I do think it then is still a reception (corruption) or buffering (presumably related to speed) issue. I suggest:
a) modifying the firmware to give you traces around packet buffer usage and signals (to detect buffering issues)
b) log packets before writing to PCAP on the host and print packet corruption info to see if a pattern there can be identified and fixed host side

grygoryan1989, 2019-07-03:
Hello!
Just bought Api-Mote v4 out of the box does not work with zbstumbler and most of the KillerBee tools. Judging by the questions raised here, the problem is not only mine, but there is no answer how to remove errors and ensure normal work. Do I understand correctly that it's about the firmware, and I need to wait for its update? Or maybe another solution?
Symptoms are standard:
When zbstumbler is running, the output is "debug clearing owerflow", in wiresharke there is "bad FCS". As far as I understand, these are components of one problem.
Thank!

rmspeers, 2019-07-03:
Just to sanity check: What is your status in zbid? Do you get any frames?

It sounds like you're seeing the same issues discussed above though, and my answers above are probably applicable.

grygoryan1989, 2019-07-03:
Thanks for the quick response!
zbid prints api mote v2, /dev/ttyUSB0, as it should be.
No answers have been given above on how to solve these problems, except for a certain “pairing mode”, but it is unclear how to configure it or how to enable it. No other solutions are given. Please forgive me if I do not understand something - I am a beginner in this topic.
The equipment under test is standard industrial equipment; in addition, work on equipment manufactured by different manufacturers was tested with the same result.

rmspeers, 2019-07-03:
I don't see a reference to pairing mode above. To overcome these issues, someone will probably need to do firmware development as described in https://github.com/riverloopsec/killerbee/issues/173#issuecomment-507081850.

rmspeers, 2020-06-07:
Any updates here? Will close due to inactivity otherwise.

closed 2020-07-23 by rmspeers