riverloopsec/killerbee #174: Zbstumbler not detecting beacons

amam76, 2019-07-07:
I have Philips Hue smart Hub v1 and Philips smart bulb. I am using API Mote and zbstumbler as below

Setting channel to 12.
Transmitting beacon request.
Setting channel to 13.
Transmitting beacon request.
Setting channel to 14.
Transmitting beacon request.
Setting channel to 15.
Transmitting beacon request.
# DEBUG Clearing overflow
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.
Received frame is not a beacon (FCF=4188).
Received frame.
Received frame is not a beacon (FCF=4188).
Setting channel to 16.
Transmitting beacon request.
Received frame.
Received frame is not a beacon (FCF=4188).
Setting channel to 17.
Transmitting beacon request.

I was told before that I need an antenna to get this working. I got an antenna as in the attached image but nothing has changed and I still can't get APIMote working.

Any advise is appreciated.

Regards,

rmspeers, 2019-07-07:
Good to have an antenna on. As documented in the FAQ (https://github.com/riverloopsec/killerbee/blob/master/FAQ.md#does-not-get-frames-received), you need to have an antenna on unless you switched the position of the antenna selector passive.

So, have you looked at the received frames in Wireshark or with Scapy? Is the frame received a beacon? You can see that zbstumbler is one of the older KillerBee tools, and doesn't use Scapy internally currently, so it just does a mask check on the FCF at https://github.com/riverloopsec/killerbee/blob/master/tools/zbstumbler#L73. The masks are in https://github.com/riverloopsec/killerbee/blob/master/killerbee/dot154decode.py#L7 IAW  the spec. So, what is the frame you are getting? Is it _really_ an 802.15.4 beacon? What's the hex of the full frame, and a PCAP showing what is happening on the channel while this all happens?

amam76, 2019-07-08:
I had RZ Raven Before and it worked perfectly with zstumbler (until I lost it). 
I didn’t have to get the hex of the full frame or analyze the pcap. It just worked!
From the number of similar issues raised by other users I would question the reliability of this tool.
Maybe it was a mistake to buy this tool.

rmspeers, 2019-07-08:
I'm happy to try to assist if you're able to dig in and analyze what's going wrong. However, I'm unable to help you use the tool remotely if you are unable to assess the underlying traffic and troubleshoot it. As I'm sure you noticed, KillerBee is labeled as something to be used by people with an understanding of the underlying protocol and due to the joy of RF, it doesn't always just work out of the box. Further, as I'm sure you noticed you're using a v4_beta_ hardware dongle, that is a beta by it's name and has documented edge cases. Having said that, myself and many others use it daily including with zbstumbler with success.

mswara0, 2020-03-06:
Hi,
I have recently purchased api mote v4 beta. I am a beginner. i have installed killer bee in kali2 as instructed in readme. when i typed zbid,it gives the below output.
root@kali:~# zbid
           Dev Product String       Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2            
root@kali:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

the system detects the device but when i execute zbstumbler, it gets stuck and shows:
root@kali:~# zbstumbler
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'

and thats it. doesnt say any thing else.
Also, when i run zbwireshark on any channels, the wireshark window opens.
root@kali:~# zbwireshark -c 20 -i /dev/ttyUSB0
zbwireshark: listening on '/dev/ttyUSB0', channel 20, page 0 (2450.0 MHz), link-type DLT_IEEE802_15_4, capture size 127 bytes

nothing appears on window. i am not sure what causing this. please help me with this.
Thankyou.

riverloopsec, 2020-03-08:
Glad to hear the device seems to be enumerating and connecting fine to your
host.

You can try the verbose flags to see more about what zbstumbler is doing,
but I would first suggest double checking the frequencies and channel of
the device you are looking for. Depending on the type of device, while
running zbwireshark you should be able to see packets appear as you do
something like turn on/off a light. If you are not seeing that, you need to
make sure you have a connected antenna on the apimote, check your RF
channel and frequency, etc.

mswara0, 2020-03-09:
Hi,
thank you for replying. My initial setup is two xbee s1 (one router and other is coordinator works on 2.4ghz) connected and router sending same frame in loop to coordinator. these two xbees are connected on different laptop. on xctu I can see the transmitted and recieved packets.

I have other system, with kali2 and ubuntu 12 installed on virtual box.
as you said, I ran zbstumbler wilth -v. it scanned all the channels (11 to 26) bud didnot detect anything (on both os).  
then I ran wireshark on different channels one by one. i finally got those packets on channel 12 and could see the plain text  (I guess xbee does not encrypt msgs). still, zbstumbler didn't work.
 i think my issue was that i was not using antenna. 
now my next step is inject something in to the channel. I dont have any idea how to do that. if you could point me in that direction,that would be great.

thank you so much for the help.

rmspeers, 2020-03-10:
Hello,

Be aware that zbstumbler is looking for standard ZigBee beacon frames, as it is documented. Note that XBees I believe don’t quite follow that standard, if I recall correctly. I suggest you look at how they respond to a beacon request against the code and see if changes are needed.

For injecting, zbreplay or zbscapy are both ways to do this, among scripting whatever yourself.

taylorcenters, 2021-07-01:
Original issue has been resolved. Thanks

closed 2021-07-01 by taylorcenters