riverloopsec/killerbee #176: Cannot flash api-mote

Staffolainen, 2019-07-25:
Hi

I cannot flash my newly purchased ApiMote v4. I tried to flash it to solve issues with clearing overflow because I haven't been able to sniff any packets.

I get this error message
```
parallels@parallels-vm:~/killerbee/firmware$ sudo ./flash_apimote.sh 
Flashing apimotev4_gf.hex to the connected USB serial device.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "./goodfet.bsl", line 1982, in <module>
    main(0);
  File "./goodfet.bsl", line 1882, in main
    bsl.actionStartBSL()
  File "./goodfet.bsl", line 1164, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "./goodfet.bsl", line 1134, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "./goodfet.bsl", line 800, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "./goodfet.bsl", line 479, in comTxRx
    rxHeader, rxNum = self.comRxHeader()        #receive header
  File "./goodfet.bsl", line 385, in comRxHeader
    if not hdr: raise BSLException("Timeout")
__main__.BSLException: Timeout

```

Runs on Ubuntu 16.04 (clean machine) in parallells on a MacBook Pro.

Any suggestions?

BR
Staffan

Staffolainen, 2019-07-25:
After som tries with removing --speed option and trying from a RPI I finally succeeded with download but the device doesn't present itself to zbid. Instead it gives this error message
**See the GoodFET FAQ about missing info flash.**. Any ideas in where the problem is?

`pi@raspberrypi:~/killerbee/firmware $ board=apimote3 ./goodfet.bsl -e -p 'apimotev4_gf.hex'
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "./goodfet.bsl", line 1982, in <module>
    main(0);
  File "./goodfet.bsl", line 1882, in main
    bsl.actionStartBSL()
  File "./goodfet.bsl", line 1164, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "./goodfet.bsl", line 1134, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "./goodfet.bsl", line 800, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "./goodfet.bsl", line 479, in comTxRx
    rxHeader, rxNum = self.comRxHeader()        #receive header
  File "./goodfet.bsl", line 385, in comRxHeader
    if not hdr: raise BSLException("Timeout")
__main__.BSLException: Timeout
pi@raspberrypi:~/killerbee/firmware $ board=apimote3 ./goodfet.bsl -e -p 'apimotev4_gf.hex'
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Program ...
7874 bytes programmed.
pi@raspberrypi:~/killerbee/firmware $ zbid
           Dev Product String       Serial Number
**See the GoodFET FAQ about missing info flash.**
pi@raspberrypi:~/killerbee/firmware $ `

rmspeers, 2019-07-25:
It looks like the flashing was done successfully in the above paste.

You may want to try a GoodFET tool like ./goodfet.monitor info or similar to communicate to the MSP430, but things look like they are probably working and you may just be hitting the beta issue where you need to try connecting a few times (zbid rerun till it gets a serial sync).

Staffolainen, 2019-07-25:
Hallelujah! You were right! Now zbid finds the device and it works as intended. I tore my hair whole afternoon with this issue with retries to flash the device and googling. Thanks.

BR
Staffan

closed 2019-07-25 by Staffolainen