riverloopsec/killerbee #177: zbid output "See the GoodFET FAQ about missing info flash."

Staffolainen, 2019-07-26:
The API-mote unstable and zbid returns 
`See the GoodFET FAQ about missing info flash.` about 3 out of 4 times. zbdump and/or zbwireshark returns `DEBUG Clearing Overflow` most of the time.

Is there any communication tweaking that can be done to get this stable/working?  

Same result with different cables and different machines. Runs Ubuntu 16.04 in parallels in OSX and Raspbian on a RPI 3b.

Staffolainen, 2019-07-26:
The reason I need this working is that we're gonna demo IoT insecurities for our customers on event in September there a replay attack on some hue lamps are planned to be included so this must work every time (its enough with the demo ghost!:) ). Is there any other tranceivers that works better with killerbee than ApiMote and are less experimental/beta? In that case I can return this one and purchase something else.

RobertoD91, 2020-06-07:
Same issue, tested with attify vm (ver 3 and 2)

rmspeers, 2020-06-07:
The missing info flash message is within the GoodFET code which is used to connect to it - I will defer to @travisgoodspeed but I believe it is mostly harmless, or occurs when serial communication via the FTDI chip doesn't get established?

As I've suggested in other threads, debugging of this at the FTDI and USB level is my best suggestion, possibly to include programming the FTDI chip to non-default settings to make it work better.

Others often have luck using the `-i` flag to specify a device to not make it enumerate every time which can cause issues.

Also, as mentioned before in other threads I tend to have good behavior on Ubuntu not in a VM, and 12.04 (although now very old...) I recall at being very very stable with it.

If you are looking for other receivers, see the docs in the repo which lists features of a number of them, and you could also add injection support to one of those firmware and contribute it if you need that. ApiMote is a beta device due to this instability and is meant for deep advanced research at the PHY/MAC layers.

closed 2021-06-22 by taylorcenters