riverloopsec/killerbee #178: Add tshark support for headless sniffing support

twistedvip, 2019-08-16:
This solution is very valuable for some zigbee testing on a Linux machine, however, I see zbwireshark currently is hard-coded to use wireshark which makes this unable to work when wanting to run automated headless testing (say through Jenkins).

From initial tests, it looks like it should be an easy modification of zbwireshark of simply modifying the start_wireshark() function changing 'wireshark' to 'tshark', removing '-k' argument, and adding an output file argument, giving the location to tshark with the '-w' argument.

A possible solution could simply be a new tool called 'zbtshark'.

rmspeers, 2019-08-16:
Sure,  do you have a PR you’d like to submit for this? If not, I will keep this ticket open as an idea to get to in an upcoming release.

twistedvip, 2019-08-26:
@rmspeers Actually trying to create my own branch, I'm not sure I have permissions to do so and create a pull request. I'll just attach the zbtshark file here. I haven't really done much testing, just wanted to give the general idea for you. (Had to zip the file up since github only supports a limited amount of file formats.

[zbtshark.zip](https://github.com/riverloopsec/killerbee/files/3541264/zbtshark.zip)

taylorcenters, 2021-06-22:
hi, you'll need to fork the repo and create the branch and changes you want, then you can open a PR from that fork back into this repo. From there we will be able to work with you while reviewing the PR

closed 2021-06-22 by taylorcenters