riverloopsec/killerbee #182: Unable to flash API Mote with GoodFET

sandwich-destroyer, 2019-09-03:
Please see https://github.com/travisgoodspeed/goodfet/issues/46

sandwich-destroyer, 2019-09-04:
For reference here is the error I was getting pre-re-flash:
```
root@kali:/opt/killerbee/tools# python zbstumbler -v -i /dev/ttyUSB0
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
Setting channel to 12.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
issue in peeking for a register
list index out of range
ERROR: Unable to inject packet: list index out of range
root@kali:/opt/killerbee/tools# python zbid
           Dev Product String       Serial Number
See the GoodFET FAQ about missing info flash.
```

Here is the error I am getting with the flash_apimote.sh script. 
```
root@kali:/opt/killerbee/firmware# ./flash_apimote.sh 
Flashing apimotev4_gf.hex to the connected USB serial device.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Changing baudrate to 38400 ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "./goodfet.bsl", line 1982, in <module>
    main(0);
  File "./goodfet.bsl", line 1882, in main
    bsl.actionStartBSL()
  File "./goodfet.bsl", line 1164, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "./goodfet.bsl", line 1134, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "./goodfet.bsl", line 800, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "./goodfet.bsl", line 479, in comTxRx
    rxHeader, rxNum = self.comRxHeader()        #receive header
  File "./goodfet.bsl", line 385, in comRxHeader
    if not hdr: raise BSLException("Timeout")
__main__.BSLException: Timeout
```

Please advise :-)

rmspeers, 2019-09-04:
If you want to reflash, try the command in flash_apimote.sh without the --speed flag, and see how that works for a reflash.

Also what kernel/etc are you on, and are you in a VM or on direct bare metal?

sandwich-destroyer, 2019-09-04:
Thank you. Looks like the flash worked, I removed the `--speed` argument from the script. 
```
root@kali:/opt/killerbee/firmware# ./flash_apimote.sh
Flashing apimotev4_gf.hex to the connected USB serial device.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Program ...
7874 bytes programmed.
root@kali:/opt/killerbee/firmware# uname -a
Linux kali 4.19.0-kali5-amd64 #1 SMP Debian 4.19.37-5kali1 (2019-06-20) x86_64 GNU/Linux
```
I am using a VM.

sandwich-destroyer, 2019-09-04:
Still getting the strange `missing info flash` error..
```
root@kali:/opt/killerbee/tools# python zbid
           Dev Product String       Serial Number
See the GoodFET FAQ about missing info flash.
```

sandwich-destroyer, 2019-09-04:
I am now trying it on metal, I am still getting the same results. The flash_apimote.sh script fails with a timeout, then after running again successfully programs 7874 bytes and killerbee is still unable to find it.

```
root@bench-1:/opt/killerbee/firmware# goodfet.bsl -e -p apimotev4_gf.hex 
Board not specified.  Defaulting to goodfet41.
Press Ctrl+C to cancel, or Enter to continue.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Traceback (most recent call last):
  File "/usr/local/bin/goodfet.bsl", line 1987, in <module>
    main(0);
  File "/usr/local/bin/goodfet.bsl", line 1887, in main
    bsl.actionStartBSL()
  File "/usr/local/bin/goodfet.bsl", line 1166, in actionStartBSL
    self.txPasswd(self.passwd)                  #transmit password
  File "/usr/local/bin/goodfet.bsl", line 1136, in txPasswd
    wait=wait)           #if wait is 1, try to sync forever
  File "/usr/local/bin/goodfet.bsl", line 801, in bslTxRx
    rxFrame = self.comTxRx(cmd, dataOut, len(dataOut))  #Send frame
  File "/usr/local/bin/goodfet.bsl", line 480, in comTxRx
    rxHeader, rxNum = self.comRxHeader()        #receive header
  File "/usr/local/bin/goodfet.bsl", line 386, in comRxHeader
    if not hdr: raise BSLException("Timeout")
__main__.BSLException: Timeout
root@bench-1:/opt/killerbee/firmware# goodfet.bsl -e -p apimotev4_gf.hex 
Board not specified.  Defaulting to goodfet41.
Press Ctrl+C to cancel, or Enter to continue.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Program ...
7874 bytes programmed.
root@bench-1:/opt/killerbee/firmware# python ../tools/zbid
           Dev Product String       Serial Number
See the GoodFET FAQ about missing info flash.
root@bench-1:/opt/killerbee/firmware# uname -a
Linux bench-1 5.0.0-27-generic #28~18.04.1-Ubuntu SMP Thu Aug 22 03:00:32 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

Dec0y-jb, 2019-10-01:
Hello, I am having the same issue. I was having some issues with the API-Mote originally and thought reflashing the firmware might help. Now I am worse off than before...

`~/killerbee/firmware$ sudo bash flash_apimote.sh 
Flashing apimotev4_gf.hex to the connected USB serial device.
MSP430 Bootstrap Loader Version: 1.39-goodfet-8
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Checking for info flash...  None.
Look at contrib/infos/README.txt for better performance.
Mass Erase...
Transmit default password ...
Invoking BSL...
Transmit default password ...
Current bootstrap loader version: 2.13 (Device ID: f26f)
Program ...
7874 bytes programmed.
`
`~/killerbee/firmware$ sudo zbid
           Dev Product String       Serial Number
See the GoodFET FAQ about missing info flash.
`

chewwie100, 2020-02-28:
Another person in the same boat here. I attempted to change the board type around, but to no avail. Any solution? Regretting attempting to reflash the board now, haha.

rmspeers, 2020-06-25:
Just as a note, swap retest test should only be for apimote v1, not any later HW versions.

rmspeers, 2020-07-01:
Ok, one of my colleagues just flashed 50 as a test and confirmed that removing the `--speed` command from the `flash_apimote.sh` script resolves this issue that sometimes occurs (that manifests in `__main__.BSLException: Timeout`).

closed 2020-08-05 by rmspeers