riverloopsec/killerbee #185: AVRRZUSBSTICK Production has ceased.

971106029, 2019-09-10:
Hi
      AVRRZUSBSTICK Production has ceased.        

       ATZB-X-233-USB
   Is there the same function?  
Can I load it on ATZB-X-233-USB?

rmspeers, 2019-09-10:
It appears you are asking if the device at https://www.microchip.com/DevelopmentTools/ProductDetails/PartNO/ATZB-X-233-USB#additional-summary can be used.

I have not tried one but it looks like there is at least firmware for sniffing available, packaged in their download (see https://www.avrfreaks.net/forum/sniffing-atzb-x-233-usb or 3.2.3 of http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42194-ZigBit-USB-Stick-User-Guide.pdf for directory path). You could use this firmware, and write a dev_*.py file that implements pnext() to get the data from it.

Otherwise, you could write your own firmware as they seem to provide all the development tools, and this would allow you to have additional features like injection, probably jamming, etc. See 1.4 of http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-42194-ZigBit-USB-Stick-User-Guide.pdf for where to get the example code according to them that you could build off of.

We don't have one of these devices to experiment with but it looks

971106029, 2019-09-12:
ok  ,thank you very much . 

-

closed 2020-06-07 by rmspeers