riverloopsec/killerbee #187: cannot dissect packets

mstekker, 2019-09-13:
Hi,

I am using the scapy_extensions from this great project. When using "kbsniff", all works fine, the "prn" handler receives the packets.

I would now like to extract data from the packet fields. Example (getting source MAC):

if packet.haslayer (ZigbeeNWK):
 source = packet[ZigbeeNWK].source

source now is an int. I can convert it to hex with hex(source). But... if the mac address is starting with 00 this approach fails (the first 00 will miss).

if I use packet.show () all data from the packet shows in the correct human readable presentation. I am unable to retrieve data from the packets in such a way. 

Is there an (other) option to retrieve the human readable data (like the content from packet.show()) when using statements like:  x = packet[ZigbeeNWK].source

If none, maybe a suggestion for a next release?

Thanks in advance for any suggestions.

Mart

hgbstu, 2019-09-13:
The source variable contains the logical Zigbee address, that means it's a 16 bit integer value and not a MAC address.

The MAC address is located in the ext_src variable, as long as the extended_src flag is set in the packet.

mstekker, 2019-09-13:
hi hgbstu,

Thank you. You are right, that was not the MAC address but my question stays the same. Example:

packet[ZigbeeSecurityHeader].show () 

##[ Zigbee Security Header ]### 
reserved1 = 
extended_nonce= 1
key_type = network_key
nwk_seclevel= None
fc = 0x7ce1739
source = 00:17:88:01:01:aa:01:0f
key_seqnum= 0
data = '*j6\xc0\x9f .|\xf6\x9a\x81i\x86)$\x05Y\x0e\xeb'
mic =

As you can see, .show () will result in nice formatted data.

Now I use:
mac = packet[ZigbeeSecurityHeader].source

will result in an int

<type 'int'>

I can convert this int into hex with hex(mac)

and get:

0x17880101aa312f

the leading 00 is missing (which is understandable).

Now my question is, is there a way to access fields in a way packet.show () does and thus get the human readable and correct values.

Thank you again.

Mart

rmspeers, 2019-09-13:
You want to call the scapy function `i2h` (I believe) on the field you get. So, extract the field (I believe scapy calls this `get_field('source')` or something), and then try calling i2h (which it should have defined) on it. That is what `show()` is doing internally.

mstekker, 2019-09-13:
Hi,

Thanks for the explanation. I did not know i2h. Started digging into it. 

mac = packet[ZigbeeSecurityHeader].get_field ('source').i2h ()

Generates an exception. "i2h" needs 2 parameters (or I am doing something wrong)  which and how I will try to find out.

Thanks!

mstekker, 2019-09-14:
Found no solution so far. An alternative would be to parse the result of packet.show (). Not the clean way, but as a workaround.

Mart

closed 2019-09-14 by mstekker