riverloopsec/killerbee #192: Api Mote Zbdump capturing packets with Bad FCS

fokkarwai, 2019-10-15:
I am currently using the ApiMote to sniff traffic between a Philips Hue Bridge and a Philips Hue lightbulb. I have also attached an external antenna to the ApiMote.

Using zbdump, I am able to capture the packets transmitted by the devices. However most the packets captured have the Bad FCS issue and are not interpreted as Zigbee packets. I have read the replies to some of the related issues and it was raised that the issue could be due to corruption of packets due to reception problem. However, for my case I am able to observe that the broadcast packets are correctly captured as Zigbee packets. On the other hand, all of the point to point packets between the bridge and the bulb have the Bad FCS problem. Since there is a very specific pattern where the problem occurs, is it still a reception problem? I have attached the PCAP file for reference.

Also I am having the "Received frame is not a beacon" issue when using zbstumbler. How am I able to capture and observe the packets transmitted at the same time when running zbstumbler?

[test.zip](https://github.com/riverloopsec/killerbee/files/3728431/test.zip)

llcoder00, 2020-02-23:
I received my attify atimote (flashed with killerbee) yesterday and regrettably, I'm seeing the exact same thing, regardless if I use an external antenna or not. There appears to be many examples of similar experiences, yet there doesn't seem to be any real solution. Frustrating.

riverloopsec, 2020-02-23:
Most people have found that it’s solved with distance and or slower packet
speed or less interference.

Of course bad FCS can be unrelated to capture as well and just be an
interference issue but in most cases it seems to be more than that - and
also be impacted by the speed the current firmware can handle incoming
packets at.

If someone is looking to improve it I would suggest working in the firmware
to improve the buffering of packets and the speed of the loop that reads
them off the radio.

rmspeers, 2020-06-07:
Any updates here?

llcoder00, 2020-06-07:
Thanks for checking in. Unfortunately, I've had to switch gears and move on
to an unrelated project. Soon, I will return to my Zigbee research and will
keep you in the loop. The one thing I haven't tried is isolating my Zigbee
hub to an area unaffected by possible wifi interference.Thanks again.

taylorcenters, 2021-06-25:
We believe a fix for this has been merged - please try again and reopen if you still have issues! Thanks.

closed 2021-06-25 by taylorcenters