riverloopsec/killerbee #195: Python3 Support

SecureAB, 2020-01-02:
Hey,

I currently have time to improve the Python3 support. I've seen that there was already work done in #184 and in [feature/py3](riverloopsec/killerbee/tree/feature/py3), but I still have problems with some functionaility.

Where should the Python3 development should take place? Should I rebase [feature/py3](riverloopsec/killerbee/tree/feature/py3) and write patches against this branch? Or should I first work as much as possible on the development branch? The changes like `print "1234"` to `print("1234")` can be made without problems while keeping Python2.7 compatibility.

I would prefer to start a new branch and cherry pick some changes from the py3 branch and start working there. But I'm also open to discuss my approach with you.

rmspeers, 2020-01-02:
Thanks for the help!

Yes, feel free to do a new one as feature/py3 has aged, so although I'd probably update it, you should do whatever you think is best and your preferred approach.

Let me know if you run into issues -- thank you for helping update the support to py3!

SecureAB, 2020-01-02:
Thank you for your fast response. I have seen that e.g. d91655041e5b36924497190096e2e3c9af8e03a0 is a change which was only made in the py3 branch, but it is not Python3 specific.

Should I first collect such changes and create a merge request with them against the development branch?

Is it enough, if the development branch is Python 2.7 compatible or do you want to keep support for lower python versions?

riverloopsec, 2020-01-02:
2.7 is all we need to keep support for IMO. I’ll look at that specific
commit and others soon to answer the rest of your question.

rmspeers, 2020-01-02:
Yes, the commit you found -- https://github.com/riverloopsec/killerbee/commit/d91655041e5b36924497190096e2e3c9af8e03a0 -- should indeed be in mainline.

We can handle that as a separate ticket or you can include in your update, if you want. Let me know.

SecureAB, 2020-01-04:
I collected the commits which I think should be mainline in #196 

Do you want to keep the branches with Python3 support separate from the others? It is possible to write Python 2 and Python 3 compatible code, so this is not a must.

rmspeers, 2020-01-04:
Absolutely we should merge down to a single branch with py2.7 and py3.5 (or 3.6?)+ support. I had started the py3 branch to get code that was compatible with both before merging down, and just didn't finish.

Thanks for making https://github.com/riverloopsec/killerbee/pull/196. Merged.

SteveJM, 2020-01-08:
I've submitted a PR for this (sorry, it was my Christmas project).

SecureAB, 2020-01-08:
That's not a problem @SteveJM. Have you already tested your changes with hardware and which hardware have you used? I have an ApiMote and a TelosB, so I can also test some parts if you haven't already done it for these two hardware pieces.

Some changes like the renaming of `raw_input()` to `input()` between py2.7 and py3 require either a distinction between the Python version like
```python
import sys

if sys.version_info[0] == 3:
    input()
else:
    raw_input()
```

or the use of a compatibility library like [Six](https://pypi.org/project/six/) or [Future](https://pypi.org/project/future/). I personally would prefer using the Future library.

What do you think about that @rmspeers?

rmspeers, 2020-01-08:
I'd love to minimize third party dependencies if we can, but also agree we don't want a lot of switch statements for little things like this.

In the case of raw_input, I'd be happy to just refactor that out (and can do it quickly if desired and submit a PR). But, there are probably other cases you have found?

SteveJM, 2020-01-08:
Ahhh, so for my branch I chose to support Python3 *only*.

rmspeers, 2020-01-09:
I'd happily include a python3 only version in a branch (e.g., PR to feature/py3), but would prefer one that supports py2.7+ and py3 to merge into the main branch and have everyone on a unified branch - as I think @SecureAB was planning.

rmspeers, 2020-01-09:
Also, it looks like @SecureAB had tried https://github.com/riverloopsec/killerbee/pull/184 and found some issues there -- would be great to get those resolved in some way so we can merge the Sewio updates made in that PR.

gezaba, 2020-01-23:
Hey, first of all thank you for your work!
I pulled your branch because I wanted to use the tool zbdump.
I used the Atmel RZ USB-Stick and the CC2531 USB Dongle as hardware.

Unfortunately it is not possible to get any packages with zbdump even though the program seems to work normally. When executing zbdump with python2 all works normally and I get the packets.
Maybe someone can help me and also verify this. Yes, I could use it with python2, but I need it to work with python3 support for a study project. 
thank you

SecureAB, 2020-01-26:
My problem was that the Telos-B and the Apimote USB Dongle wasn't working in #184, since the author tested only the Sewio stuff. And I also didn't liked that multiple changes (the python3 support and the Sewio updates) are done in one merge request (and even in the same commit).

SecureAB, 2020-01-26:
@gezaba You can if #198 or #184 works for you. If not, you have to either wait until we have completely ported killerbee to python3, or you have to help us doing that. It would be totally enough, if you only port the parts to python3 which your study projects needs.

gezaba, 2020-01-26:
@SecureAB thanks for your answer.
I tried all the branches but zbdump is still not capturing packets.
Before that I also tried on my own and of course keep migrating the neccessary files to python3 with the help of 2to3 tool and some little manually changes. 
Unfortunately all approaches resulted the same. No packets could be captured with the corresponding hardware just mentioned.

rmspeers, 2020-01-26:
Are you getting any warnings from pyUSB? Is zbid correctly identifying your devices in the py3 branches?

rmspeers, 2020-01-27:
CC @taylorcenters

SecureAB, 2020-02-15:
@rmspeers Is there anything you need to merge #200?

gezaba, 2020-02-15:
I wanted to let you know, that after all it was possible to use zbdump and zbstumbler with the RZ Atmel stick in python3. But with CC2531 I was definitely not able to capture traffic with zbdump in python3 although it was possible with python2 there.
However thanks for your replys and sorry for my delayed answer.

rmspeers, 2020-03-10:
@gezaba  can you try using the PR branch at https://github.com/riverloopsec/killerbee/pull/204#pullrequestreview-372278782 ?

rmspeers, 2020-04-15:
@taylorcenters  says he has a branch where he's further improved it and could use debugging help and testing - Taylor - pls share here so people can help look?

taylorcenters, 2021-04-15:
Python3 support has been merged! Thanks everyone for all the help on this effort. Please reach out if you run into problems or have any questions.

closed 2021-04-15 by taylorcenters