riverloopsec/killerbee #201: Raspbee

snylee, 2020-02-09:
Hi Guys

I'm kinda new to this so please bare with me.
I'm a student in my final year of university. For my final year project I have chosen to focus on the penetration of Zigbee networks/devices. 
So far I have purchased a Raspberry Pi with the Raspbee Premium module. I am able to see traffic through Wireshark using software from Bishop Fox's software, ZigDiggity. But i want to do more, which is how I've come across Killerbee.
I've noticed that the hardware requirements for Killerbee do not include that of Raspbee. Do you know if there is anyway I could use Killerbee with my Raspbee module? I only ask this because most of the hardware required for Killerbee are no longer in stock, especially the Atmel RZUSB Stick. I would need to sniff and inject.
Any ideas would be mostly beneficial.

Thanks Guys
Simon

rmspeers, 2020-02-09:
Yes, you should be able to add Raspbee support relatively simply -- copy the pattern in another dev_*.py file and implement those functions that are in there to speak to the Raspbee hardware via it's serial interface. Your implementation likely could be informed by looking at `zigdiggity/radios/raspbee_radio.py` or you can implement using the Raspbee specifications.

snylee, 2020-02-09:
Thanks for the quick reply rmspeers. 
So please let me know if i've got this right. 
-create a new .py script called dev_raspbee.py -(for example)
-copy the script from "zigdiggity/radios/raspbee_radio.py" into my new dev_raspbee.py script?
I'm a bit confused. Ha

rmspeers, 2020-02-09:
You will want to create dev_raspbee.py, and can base it on dev_apimote.py as an example. However, you can’t use copy something in - you will need to implement those functions (pnext, set_channel, etc) based on the Raspbee communication protocol (which, as I mentioned,  "zigdiggity/radios/raspbee_radio.py” may be one reference for how to do if you wanted to read it).

You’ll need to implement those methods and debug from that point to ensure you’re getting all the data in the expected format (e.g., FCS prepended vs null’ed, etc).

snylee, 2020-02-10:
Damn. So I’m not a programmer so i have no clue how to implement it across or what to look for in the zigdiggity radios file. Any other ideas to add support? Or do you have any idea of where i could get one of the supported hardware radios from?

rmspeers, 2020-02-10:
Code needs to be written for there to be support for a radio. I've started a branch, tagged 201-raspbee-support, but have not had time to write most of it or test it - it serves only as an example to be built upon and tested.

rmspeers, 2020-02-10:
For supported radios, you should be able to get the SiLabs Node Test supported boards from DigiKey and similar distributors easily, if those support the features you need.

snylee, 2020-02-15:
Ok thank for. I appreciate your time and effort. In the meantime time trying to do a quick corse on phthon. I know the basics but im not there yet. 😁

wang70880, 2021-01-25:
How's the status of Raspbee firmware development?

rmspeers, 2021-01-25:
At this time, I have not had time to dedicate to writing a Raspbee firmware. We welcome any contributions of this firmware.

wang70880, 2021-01-25:
I would like to help develop the Raspbee firmware. Do you have any idea to help quickly get started? I plan to first read your codes of implemented firmwares for other devices, and imitate them according to the docs of Raspbee.

rmspeers, 2021-01-25:
See above notes on https://github.com/riverloopsec/killerbee/issues/201#issuecomment-583901216 on what you need to do for the KillerBee "driver" layer. You can use the default firmware if you wish, see above considerations. `dev_raspbee.py` is, as you say, the abstraction layer for speaking to the firmware. I would start with implementing that to talk to the default firmware and you can update firmware if you find things are missing after doing that which you need.

wang70880, 2021-01-26:
> See above notes on [#201 (comment)](https://github.com/riverloopsec/killerbee/issues/201#issuecomment-583901216) on what you need to do for the KillerBee "driver" layer. You can use the default firmware if you wish, see above considerations. `dev_raspbee.py` is, as you say, the abstraction layer for speaking to the firmware. I would start with implementing that to talk to the default firmware and you can update firmware if you find things are missing after doing that which you need.

Thanks, and it seems that my understanding is correct. The last question is that I don't know how Zigdiggty found that CMD_SET_CHANNEL = 0x35 or CMD_RX = 0x31. To be honest, I didn't find any documentation about the communication protocol with Raspbee firmware. Do you have any idea?

wang70880, 2021-01-27:
However, I found that the firmware of Raspbee is not open-sourced, and the functionality of original firmware is limited, e.g., changing PANID. What should we do to solve this firmware problem?