riverloopsec/killerbee #202: Zbid: Interface not found on Debian GNU/Linux

Sandb0xED, 2020-02-10:
Hi, I'm running into an issue wherein I can't see the APIMote board as an interface but it is connected through the connector to my Debian Linux VM (running the most recent kali VM from Offensive Security). 
I read on another forum to try out alternatives: zbid -i <IP Address> or zbid --dev <serial device entry in /dev>. The former gives me an error of kbutils.devlist has an unknown type of IP Sniffer. Whereas the latter, gives me no output. 
I wonder if this has anything to do with the serial baud rate (maybe the USB not being able to talk to my machine through the serial interface). I'm really not sure. Hope this gives you enough context!

riverloopsec, 2020-02-11:
> I read on another forum to try out alternatives: zbid -i or zbid --dev

This sounds - without looking at it myself - like a typical error of the
USB to serial FTDI chip not playing well with certain USB kernel support.
This also often happens in VMs depending on the hypervisor. E.g., why the
ApiMote v4beta is still marked as a beta. I suggest trying on a bare metal
Linux machine on an older kernel. Also sometimes you can talk to it via
multiple attempts at running zbid but that is hacky!

Make sure config.py has ApiMote support turned on and disable other serial
devices there and reinstall, which is another idea but not sure that alone
will fix this.

Sandb0xED, 2020-02-11:
In this process, I tried reflashing the firmware (from your repo) onto the device a couple of times. There were time out errors, but I figured out the error thereafter. I don't know if that had anything to do with this. Is there a way for me to reset the device, so I can flash it properly for the first time?

rmspeers, 2020-02-11:
You can utilize the goodfet.bsl tooling to mass erase as needed. Flashing issues may be due to `--speed` perhaps, or due to incorrect jumper position. In general I do not suggest reflashing these devices from how they come tested/flashed.

rmspeers, 2020-06-07:
Any updates here?

Sandb0xED, 2020-06-12:
Hi there, I ended up investing in another board. Thank you for your time.

closed 2020-06-12 by Sandb0xED

inode-, 2020-11-18:
@Sandb0xED can you tell us which board did you got?