riverloopsec/killerbee #203: Killerbee - RZUSBSTICK - ZBstumbler and some other commands not functioning

JoeAJLaw, 2020-02-21:
Hey all, 

I am working on a university project to do with IoT devices, and wanted to use Killerbee in my project. However, upon receiving the RZUSBSTICK which is flashed with Killerbee which I have checked through forums and the "ZBID" command, but some commands are not appropriately functioning.  Below are some examples of the issue:

[image]

[image]

Has anyone else experienced this? Or can provide a step-by-step solution of what needs to be done?

rmspeers, 2020-03-10:
I have not seen that behavior, but although Pi3 w/ Kali 64 should work, it may be the platform

Also please confirm your Killerbee version that you installed from scratch, not using what comes on Kali.

rmspeers, 2020-06-07:
Please let me know if you have this issue anymore or can provide the above info.

grf240, 2020-06-25:
I am having a similar issue to @JoeAJLaw with using the some of the commands. I receive "Exception: Error: Semantical Error".

taylorcenters, 2021-07-01:
Hi please try again on the latest software as we've upgraded to Python3 and performed a lot of maintenance since this issue was last active.

taylorcenters, 2022-03-08:
Closing inactive - please feel free to reopen this or create a new issue if this persists with the latest software!

closed 2022-03-08 by taylorcenters