riverloopsec/killerbee #205: Problem with Silabs USB adapter

pmkohn, 2020-04-20:
Hi all,

i'm trying to use an telegesis/silabs etrx357 zigbee module with the recommended nodetest firmware. Sadly it is not working with zbwireshark:

zbwireshark -c 24 -i /dev/ttyUSB0 
Traceback (most recent call last):
  File "/usr/local/bin/zbwireshark", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.1', 'zbwireshark')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.1-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 108, in <module>
    
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.1-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 61, in main
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 171, in __init__
killerbee.kbutils.KBInterfaceError: KillerBee doesn't know how to interact with serial device at '/dev/ttyUSB0'

With minicom it responds and shows a working interface. Any hinds what i could try to debug?

rmspeers, 2020-06-07:
I'd debug in the init -- it seems like from the message that the discovery code that tries to guess what type of interface you connected is not figuring it out. Maybe add some logging in there and watch if it gets into the block that tests nodetest? Specifically, you want to look at the code around https://github.com/riverloopsec/killerbee/blob/13408a6a78aee9b480f3e6384119411d2842d5b7/killerbee/__init__.py#L147-L149.

AdamLaurie, 2020-08-31:
I just tested it with an up to date NodeTest on EFR32 and it worked fine:

Ember Node Test Application v1.0
Aug 21 2020, 09:30:26

 What's the output of your 'help' command?

taylorcenters, 2021-06-22:
closed - no response - please reopen if you have an update

closed 2021-06-22 by taylorcenters