riverloopsec/killerbee #206: Missing dependancy in Ubuntu 20.04 (Focal Fossa)

yg-ht, 2020-05-11:
Ubuntu 20.04 does not have a python-gtk2 package so this needs to be replaced (possibly inline with this repo's Python3 dev?).

Someone has created a ppa to provide it, but that means trusting someones personal repo:

[https://launchpad.net/~nrbrtx/+archive/ubuntu/python2-stuff](https://launchpad.net/~nrbrtx/+archive/ubuntu/python2-stuff)

I'm afraid I neither have the time or the skill to work this one through and I suspect more than anything else it probably signifies the end of support for Python2.

taylorcenters, 2021-06-22:
I will move zbfind ( the tool which is the sole dependency on gtk) to a new repo for posterity. This should remove the dependency.

taylorcenters, 2021-07-01:
you should be able to use now on current develop branch on Python3 without gtk/cairo dependencies. Please reopen if still having issues. Thanks

closed 2021-07-01 by taylorcenters