riverloopsec/killerbee #207: bug with scapy 2.4.3dev691

mstekker, 2020-06-21:
Hi,

Scapy changed some internal code. Now killerbee integration regarding to sniffing is broken:

`Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "./myzig.py", line 1419, in threadscan
    kbsniff (timeout=self.zig_timeout, prn=self.zig_handler, iface=self.zig_device, verbose=False)
  File "build/bdist.linux-armv7l/egg/killerbee/scapy_extensions.py", line 224, in kbsniff
    return plist.PacketList(__kb_recv(kb, count = count, store = store, prn = prn, lfilter = lfilter, stop_filter = stop_filter, verbose = verbose, timeout = timeout), 'Sniffed')

NameError: global name 'plist' is not defined
`
Any suggestions for fixing the code?

Thanks,

Mart

rmspeers, 2020-06-21:
@guedou do you happen to recognize this off the bat?

@mstekker did you happen to find the change you reference in Scapy that broke this, as that would be helpful to understand what they changed.

guedou, 2020-06-21:
I will try to do `s/plist.PacketListField/PacketListField/`.

mstekker, 2020-06-21:
> @guedou do you happen to recognize this off the bat?
> 
> @mstekker did you happen to find the change you reference in Scapy that broke this, as that would be helpful to understand what they changed.

Not yet, but I will dig into it.

mstekker, 2020-08-09:
Till now I was unable to find any solutions. @guedou any clues?

Thank you

guedou, 2020-08-09:
Did you try my suggestion? This looks like an import error from killerbee.

mstekker, 2020-08-09:
> Did you try my suggestion? This looks like an import error from killerbee.

@guedou 

Hi,

I tried, but was not able to implement it. This is the status right now:

using:
scapy_extensions.py from killerbee

with the newest scapy build will give a:

NameError: global name 'plist' is not defined 
line 224 in scapy_extensions.py

Line 224 is:

return plist.PacketList(__kb_recv(kb, count = count, store = store, prn = prn, lfilter = lfilter, stop_filter = stop_filter, verbose = verbose, timeout = timeout), 'Sniffed')

Your remark that this could be an import error sounds good. Could you give a short example how to use your solution?

Thanks!

mstekker, 2020-08-09:
Hi,

problem solved.

adding "from scapy import plist" to scapy_extensions.py will solve the problem.

Thank you!

riverloopsec, 2020-08-09:
Thanks. Do you want to submit a pull request or post a diff and I will
update in the branch?

mstekker, 2020-08-09:
> Thanks. Do you want to submit a pull request or post a diff and I will update in the branch?
> […](#)

I will do this as soon as I am home.

Thx!

taylorcenters, 2021-07-01:
This has been fixed. Thanks

closed 2021-07-01 by taylorcenters