riverloopsec/killerbee #208: Issue running commands

grf240, 2020-06-25:
[image]

Installed all dependencies, Scapy and KillerBee, as well as having the RZUSB Stick Flashed with the KillerBee firmware, but cannot utilize. I am running a USB persistence Kali Linux and using "root" to run. ZBID works just fine and recognizes the RZUSB Stick. 

Can I get some advice as to where I might be going wrong or if there is an issue with the code?

Thanks!

rmspeers, 2020-06-25:
Yes, and zbstumbler is also recognizing it's a RZUSBstick fine. It is some issue with the USB stack.

Try either going to pyUSB 0.x branch, or else  will need to be diagnosed further on the USB item. This appears to be an issue that a number of users face on newer USB stacks. I expect there is something where calls to pyUSB need to be updated or something?

You could also try taking the snippet of code that fails and reporting it to pyUSB to see if they have any ideas or have seen this?

grf240, 2020-07-03:
@rmspeers Thank you for your advice. I attempted to install and run with an older version of pyUSB and wasn't successful. Additionally, I reached out the pyUSB github repo and they directed me back to RZUSB/KillerBee project repo. I wonder if any new releases to address this bug will be added soon.

taylorcenters, 2021-07-01:
Please try with the latest software. Thanks!

taylorcenters, 2022-03-08:
Closing inactive - please feel free to reopen this or create a new issue if this persists with the latest software!

closed 2022-03-08 by taylorcenters

Crypt0s, 2023-01-18:
For what it's worth, adding a .1 second sleep after opening the RZUSBSTICK device for inject `kb.inject()` will make this error go away on the RZUSBSTICK -- I suspect that the microcontroller needs more time to react to the request.

comeete, 2023-02-14:
Hi,Excuse-me but, Crypt0s, can you explain in more details how and where in the code can we add this time please. Thanks a lot!

comeete, 2023-02-20:
Thanks Cryt0s, it works well !