riverloopsec/killerbee #209: No usb stick available for upgrading

BoxFighter, 2020-07-11:
I use AVR Wireless Software to flash firmware in win 7. After I click Upgrade in the software, it failed flash and pop up “”No usb stick available for upgrading”. And when I plug RZUSSBSTICK in win 7,win 7 can not identity the usb device. I need help

BoxFighter, 2020-07-11:
[image]

rmspeers, 2020-07-12:
I haven't done this process, but there is relatively extensive documentation on it. I presume if you're having issues maybe try another soft load method (e.g, another tool, computer, OS, etc). If that still persists, I presume you'd need to re-load the DFU bootloader. See https://github.com/riverloopsec/killerbee/blob/develop/firmware/README.md#rzusbstick which should cover both situations.

BoxFighter, 2020-07-12:
> 我尚未完成此过程，但是有相对广泛的文档。我想如果您遇到问题，可以尝试使用另一种软加载方法（例如，另一种工具，计算机，操作系统等）。如果仍然存在，我认为您需要重新加载DFU引导加载程序。请参阅https://github.com/riverloopsec/killerbee/blob/develop/firmware/README.md#rzusbstick，其中应涵盖这两种情况。

I want to ask a question.If I want use RZUSBSTICK in windows, which driver  that I  need install ?

BoxFighter, 2020-07-23:
Finally, I choose hareware way.

taylorcenters, 2021-07-01:
Hi sorry, Windows is not officially supported, I think you would just need a usb to serial ftdi driver or something. I haven't used RZUSBSTICK yet though.

taylorcenters, 2022-03-08:
Going to close this issue. Feel free to reopen or create a new one if you have more questions! Thank you!

closed 2022-03-08 by taylorcenters