riverloopsec/killerbee #210: Confirm if RZUSBSTICK firmware Actually Jams

rmspeers, 2020-07-14:
Due to comment at https://github.com/riverloopsec/killerbee/blob/13408a6a78aee9b480f3e6384119411d2842d5b7/firmware/src/kb-rzusbstick/application/rzusbstick/air_capture.c#L512, want to check if anyone has seen the RZUSBSTICK firmware successfully jamming. Please provide input. If it is not doing this right, I want to remove the KBCapabilities for jamming on that dev so no one attempts to use it.

rmspeers, 2020-07-14:
@AdamLaurie It looks like you are the lucky commenter of that line actually so tagging you in case you know

taylorcenters, 2021-06-22:
pending response to the contrary - lets remove it from KBCapabilities

AdamLaurie, 2021-06-23:
My experience was that jamming did not work on this hardware, so I have no objection to removing it.

taylorcenters, 2022-03-08:
Capabilities set to False in PR #252 - Thanks!

closed 2022-03-08 by taylorcenters