riverloopsec/killerbee #214: Killerbee framework doesn't not work in ubuntu 18.04

BrentCC, 2020-11-05:
Hi, I'm trying to install killerbee framework and run the tool zbid in ubuntu 18.04, but it doesn't work. My ubuntu and python version as follows:

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS

Python 3.8.3

Error code as follows:
`(base) marco@ubuntu:~/scapy$ zbid
Traceback (most recent call last):
  File "/home/marco/anaconda3/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.1', 'zbid')
  File "/home/marco/anaconda3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 665, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/home/marco/anaconda3/lib/python3.8/site-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/home/marco/anaconda3/lib/python3.8/site-packages/killerbee-2.7.1-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 13, in <module>
    from killerbee import show_dev
  File "/home/marco/anaconda3/lib/python3.8/site-packages/killerbee-2.7.1-py3.8-linux-x86_64.egg/killerbee/__init__.py", line 5, in <module>
    from pcapdump import *
ModuleNotFoundError: No module named 'pcapdump'`

inode-, 2020-11-26:
Python module "pcapdump" it's missing in your installation.

Please note that Killerbee support only python 2.7 (some pull requests support python3 but currently are not in the stable branch).

taylorcenters, 2021-06-22:
closed - please try on the new python3 supported version of killerbee

closed 2021-06-22 by taylorcenters