riverloopsec/killerbee #215: apiMote not working properly after flashing

inode-, 2020-11-18:
I flashed the apiMote firmware inside the killerbee and the apiMote stopped working correctly. I tried different distro (also 12.04) on real hardware but nothing help.

Sometime the device is recognized by "zbid", but most of the times no. Also zbstumbler has big problems. 

Tryied to flash GoodFET from the original repository bu doesn't help.

I think it's related to: https://github.com/riverloopsec/killerbee/issues/190

It's possible to get the "original" firmware shipped with apiMote?

inode-, 2020-11-18:
After a lot of testing and madonne I found a workaround.

Download the GoodFET, go to into the client directory and run:

`board=apimote4 ./goodfet.monitor listapps full`

before execute any zb*commands. In some way the command will "sync" with the apimote and let it working properly. Then execute zbid of zbstumbler without problem.

rmspeers, 2020-11-18:
The GoodFET ApiMote firmware that is pre-loaded by RLS is listed in firmware/, already compiled.

The source is all available in the GoodFET project if you want to build it yourself.

Thanks for posting your workaround. I would expect that even the ./goodfet.monitor info command would sync it, but perhaps not? That's great to have a workaround to share though. That 'sync' issue is the main reason ApiMote is still listed in beta.

taylorcenters, 2021-06-25:
The change from your PR was merged! Thanks!

closed 2021-06-25 by taylorcenters