riverloopsec/killerbee #219: Unable to locate packages.

VortexGif, 2020-12-16:
I'm attempting to install killerbee on a Kali VM , when installing I get a message saying python-gtk2 , python-cairo and python-usb cannot be located. 

I assume its a python2 issue? 
Any help on getting it installed correctly would be greatly appreciated.
TIA.

rmspeers, 2020-12-16:
Could you try the py3 branch and use python3?

Otherwise you may need to look at the Kali distribution sources and update your source files to install the py2 packages. That would be to follow the distribution specific guidance for that.

Remember on Kali to fully uninstall the included KillerBee before installing the one from Git.

VortexGif, 2020-12-16:
I tried to get them by using python3-gtk2 etc , but it still couldn't find them. 

I wasn't aware of preinstalled KillerBee packages on Kali , when I try to run the preinstalled ones they don't run.

rmspeers, 2020-12-16:
For GTK, see https://python-gtk-3-tutorial.readthedocs.io/en/latest/install.html#prebuilt-packages for install guidance that points to the specific repo. Note that python-gtk and cario are not required for KillerBee to install, only python-usb is. The others are just for a few GUI tools.

For python-usb, see https://pkg.kali.org/pkg/pyusb based on a quick google.

VortexGif, 2020-12-16:
I've have usb installed and I'm still getting the same errors. I've checked pip list to see if usb is installed and it looks as if it is.
[image]

Anymore help would be appreciated a lot , really need to get this working for a university project.

VortexGif, 2020-12-16:
Fixed this issue , just used an older version of Kali. Killerbee is now lauching and working. 
I am using the CC2531 and it doesn't seem to be showing in zbid. It is showing in the connected serial ports and on lsusb.
Could this be a firmware issue?

rmspeers, 2020-12-16:
Make sure your pip and python versions are aligned, and the libusb stuff under the hood is installed on the OS. You don't just need the python layer.

For the CC2531, you'll have to check your version of firmware and debug it - start looking in the kbutils detection code probably to make sure VID/PID match, etc.

VortexGif, 2020-12-17:
are you aware of what firmware I need to use with Killerbee? I currently have the firmware that came with ZBOSS flashed on.  I can change it if I need to.

rmspeers, 2020-12-17:
I'm not sure what ZBOSS is? The one that is the standard firmware, which supports sniffing, is what is shipped by TI when one is ordered.

We are working on a separate thread for a KillerBee-specific firmware but it is still in development/test in a branch.

VortexGif, 2020-12-17:
ZBOSS is a piece of sniffing software for Zigbee packets , it's not the firmware that came on the stick.
I'll try flashing the TI firmware on the stick see if zbid picks it up.

rmspeers, 2020-12-17:
Ok, yes, if you want to support a different piece of software, then you'll need to create a new `dev_*` for it and update the detection logic, and submit a PR with that we can test. If that works, we'd update the instructions in the `firmware/README.md` to document that additional support.

VortexGif, 2020-12-17:
Right , I have the CC2531 working with Killerbee now , found some firmware buried in some folders.  Going to try the custom firmware I saw for packet injection with the CC2531.

Thanks for the help I really appreciate it , feel like I'm getting somewhere with the project I'm doing now.

taylorcenters, 2021-06-22:
Let us know if you have any more issues

closed 2021-06-22 by taylorcenters

Mizuki-ENG, 2022-10-01:
@VortexGif 
May i ask you pls, whats the version of your pyusb (apt-get or pip install) or any alternative library that can also use instead of pyusb ?
ps Can i get your custom fw thats work for cc2531 ?
ps2 Can i get the buried firmware url(link) that you mention abve ?

VortexGif, 2022-10-01:
@Mizuki-ENG 
I am not 100% sure about the version of pyusb I was using, I know I had to downgrade my kali VM to an older version to get it working, I don't have access to the VM anymore so I can't check the version used.

I can't find the custom firmware I used, It was online not written by myself. I believed I used it for packet injection.

The firmware that was buried in files came with ZBOSS, it allowed for sniffing of Zigbee traffic using windows. If you have a copy of ZBOSS it will be in ZBOSSDIR\hw\CC2351 USB Dongle

Apologies if my answer is a bit vague it's been a while since I worked on the project and my notes around it are not the best. If you have any more questions I'll try my best to answer them.