riverloopsec/killerbee #220: KillerBee doesn't know how to interact with serial device at '/dev/ttyACM0

oleyyh, 2020-12-18:
sudo zbwireshark -c 11
`<pre>Traceback (most recent call last):
  File &quot;/usr/local/bin/zbwireshark&quot;, line 4, in &lt;module&gt;
    __import__(&apos;pkg_resources&apos;).run_script(&apos;killerbee==2.7.1&apos;, &apos;zbwireshark&apos;)
  File &quot;/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py&quot;, line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File &quot;/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py&quot;, line 1469, in run_script
    exec(script_code, namespace, namespace)
  File &quot;/usr/local/lib/python2.7/dist-packages/killerbee-2.7.1-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark&quot;, line 108, in &lt;module&gt;
    
  File &quot;/usr/local/lib/python2.7/dist-packages/killerbee-2.7.1-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark&quot;, line 61, in main
    
  File &quot;build/bdist.linux-x86_64/egg/killerbee/__init__.py&quot;, line 171, in __init__
killerbee.kbutils.KBInterfaceError: KillerBee doesn&apos;t know how to interact with serial device at &apos;/dev/ttyACM0&apos;.
</pre>`

System: Ubuntu 20.04
Device: CC2531. It is recognised by the host, when using the lsusb command.
`Bus 001 Device 021: ID 0451:16a8 Texas Instruments, Inc. CC2531 ZigBee`

rmspeers, 2020-12-18:
What firmware are you running on it? Is it the firmware in the PR that was submitted of https://github.com/riverloopsec/killerbee/pull/218 ?

oleyyh, 2020-12-21:
I'm following the guide written on https://hal9k.dk/sniffing-philips-hue-zigbee-traffic-with-wireshark/ that specifies running the stock firmware on the CC2531 should work. I do however have a CC debugger that can flash CC2531 in the mail, that should arrive soon if required.

taylorcenters, 2021-06-22:
Hi - can you try again on the latest version of killerbee using python3?

taylorcenters, 2021-07-01:
Please reopen if you are still seeing this issue on the latest software. Thanks

closed 2021-07-01 by taylorcenters