riverloopsec/killerbee #226: no libraries for Kali Linux 2021.

lucianoferrari, 2021-06-11:
I cannot install the required python libraries for Kali Linux 2021. I also tried to get an old version of Ubuntu to see if it would work and it's not working properly. this is the zbid output:
 Dev Product String       Serial Number
           3:6 KILLERB001           FFFFFFFFFFFF

and when I try to use zbstumbler after a while it gives me some error messages:

iot@attifyos ~/t/k/tools> sudo zbstumbler -v
Warning: You are using pyUSB 1.x, support is in beta.
zbstumbler: Transmitting and receiving on interface '3:6'
Setting channel to 11.
Transmitting beacon request.
Traceback (most recent call last):
  File "/usr/local/bin/zbstumbler", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==2.7.1', 'zbstumbler')
  File "/home/iot/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/home/iot/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1460, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.1-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbstumbler", line 199, in <module>
    
  File "build/bdist.linux-x86_64/egg/killerbee/__init__.py", line 358, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 546, in pnext
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 385, in sniffer_on
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 355, in _open_stream
  File "build/bdist.linux-x86_64/egg/killerbee/dev_rzusbstick.py", line 334, in __usb_write
Exception: Error: Semantical Error

taylorcenters, 2021-06-14:
Can you try installing the latest version of killerbee that uses python3? It looks like you're still on the 2.7 version.

lucianoferrari, 2021-06-14:
> Can you try installing the latest version of killerbee that uses python3? It looks like you're still on the 2.7 version.

Thanks for your reply. When I try to install the new version of killerbee I'm having issues with python libraries:
┌──(kali㉿kali)-[~/killerbee]
└─$ sudo python3 setup.py install                 
[sudo] password for kali: 

Library requirements not met.  Install the following libraries, then re-run the setup script.

usb

Library recommendations not met. For full support, install the following libraries, then re-run the setup script.

gtk
You are using pyUSB 0.x. Consider upgrading to pyUSB 1.x.

The following commands should install these dependencies on Ubuntu, and can be adapted for other OSs:
        sudo apt-get install -y python-gtk2 python-usb
                                                                                                                                                                                                               
┌──(kali㉿kali)-[~/killerbee]
└─$ sudo apt-get install -y python-gtk2 python-usb                                                                                                                                                         1 ⨯
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-gtk2
E: Unable to locate package python-usb

It can't find python-gtk2 or python-usb

taylorcenters, 2021-06-14:
the gtk is only needed for the zbfind tool so you can skip that dependency if you're not using that. Also, according to https://stackoverflow.com/questions/65241073/i-cant-install-python-gtk2-on-kali-20-04 it has been removed from Kali.

for the usb library can you try apt install python3-usb?

lucianoferrari, 2021-06-14:
Thanks, I'll try that tonight. and nice to chat with you again! :-)

taylorcenters, 2021-06-14:
Let me know if that works and I'll make the PR to update the docs and link to this issue. It was good to see your name pop-up!

lucianoferrari, 2021-06-14:
Ok, thanks for the update. I had installed python3-usb successfully.
when I tried to run the setup again this is what I got:

──(kali㉿kali)-[~/killerbee]
└─$ sudo apt-get install python3-usb                                                                                                                                                                     100 ⨯
[sudo] password for kali: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  fonts-roboto-slab gir1.2-ayatanaappindicator3-0.1 gir1.2-gtksource-3.0 gir1.2-javascriptcoregtk-4.0 gir1.2-nm-1.0 gir1.2-soup-2.4 gir1.2-webkit2-4.0 gobject-introspection libical3 libjs-skeleton
  libncurses5 libtinfo5 node-normalize.css pwgen python-mpltoolkits.basemap-data python-pip-whl python3-advancedhttpserver python3-aiodns python3-aiomultiprocess python3-aiosqlite python3-aniso8601
  python3-appdirs python3-boltons python3-cairo-dev python3-dicttoxml python3-ecdsa python3-flask-restful python3-geoip2 python3-geojson python3-graphene python3-graphene-sqlalchemy python3-graphql-core
  python3-graphql-relay python3-icalendar python3-jupyter-core python3-maxminddb python3-mistune python3-mpltoolkits.basemap python3-promise python3-pycares python3-pyee python3-pyotp python3-pyppeteer
  python3-pyproj python3-pyrsistent python3-pyshp python3-requests-file python3-retrying python3-rq python3-rule-engine python3-rx python3-singledispatch python3-smoke-zephyr python3-texttable
  python3-unicodecsv python3-uvloop
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-usb
0 upgraded, 1 newly installed, 0 to remove and 69 not upgraded.
Need to get 38.9 kB of archives.
After this operation, 244 kB of additional disk space will be used.
Get:1 http://kali.download/kali kali-rolling/main amd64 python3-usb all 1.0.2-2 [38.9 kB]
Fetched 38.9 kB in 1s (43.9 kB/s)      
Selecting previously unselected package python3-usb.
(Reading database ... 299439 files and directories currently installed.)
Preparing to unpack .../python3-usb_1.0.2-2_all.deb ...
Unpacking python3-usb (1.0.2-2) ...
Setting up python3-usb (1.0.2-2) ...

                                                                                                                                                                                                           
┌──(kali㉿kali)-[~/killerbee]
└─$ sudo python3 setup.py install                 

Library recommendations not met. For full support, install the following libraries, then re-run the setup script.

gtk

The following commands should install these dependencies on Ubuntu, and can be adapted for other OSs:
        sudo apt-get install -y python-gtk2
/usr/local/lib/python3.9/dist-packages/setuptools/dist.py:484: UserWarning: Normalizing '3.0.0-beta.0' to '3.0.0b0'
  warnings.warn(tmpl.format(**locals()))
running install
running bdist_egg
running egg_info
creating killerbee.egg-info
writing killerbee.egg-info/PKG-INFO
writing dependency_links to killerbee.egg-info/dependency_links.txt
writing requirements to killerbee.egg-info/requires.txt
writing top-level names to killerbee.egg-info/top_level.txt
writing manifest file 'killerbee.egg-info/SOURCES.txt'
reading manifest file 'killerbee.egg-info/SOURCES.txt'
adding license file 'LICENSE.txt'
writing manifest file 'killerbee.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-x86_64-3.9
creating build/lib.linux-x86_64-3.9/killerbee
copying killerbee/dev_telosb.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/scapy_extensions.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/dev_zigduino.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/pcapdlt.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/GoodFET.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/dev_cc253x.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/dev_apimote.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/config.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/GoodFETatmel128.py -> build/lib.linux-x86_64-3.9/killerbee
copying killerbee/daintree.py -> build/lib.linux-x86_64-3.9/killerbee
[... 13 lines cut ...]
copying killerbee/GoodFETAVR.py -> build/lib.linux-x86_64-3.9/killerbee
creating build/lib.linux-x86_64-3.9/killerbee/openear
copying killerbee/openear/gps.py -> build/lib.linux-x86_64-3.9/killerbee/openear
copying killerbee/openear/capture.py -> build/lib.linux-x86_64-3.9/killerbee/openear
copying killerbee/openear/__init__.py -> build/lib.linux-x86_64-3.9/killerbee/openear
copying killerbee/openear/scanner.py -> build/lib.linux-x86_64-3.9/killerbee/openear
creating build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/db.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/capture.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/testGPS.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/scanning.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/__init__.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
copying killerbee/zbwardrive/zbwardrive.py -> build/lib.linux-x86_64-3.9/killerbee/zbwardrive
running build_ext
[... cut ...] /zigbee_crypt/zigbee_crypt.o
zigbee_crypt/zigbee_crypt.c:15:10: fatal error: gcrypt.h: No such file or directory
   15 | #include <gcrypt.h>
      |          ^~~~~~~~~~
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
                                                                                                                                                                                                               
┌──(kali㉿kali)-[~/killerbee]

taylorcenters, 2021-06-14:
You'll need to apt install libgcrypt-dev

lucianoferrari, 2021-06-14:
it worked now! Thank you so much. Have a wonderful week

lucianoferrari, 2021-06-14:
well
kind of.. It installed but when I run zbid this is what i got
   31 | static const char rcsid[] = "$Id$";
      |                   ^~~~~
src/_counter.c: In function ‘PyInit__counter’:
src/_counter.c:570:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  570 |     if (m == NULL)
      |     ^~
src/_counter.c:573:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  573 |  return m;
      |  ^~~~~~
build/bdist.linux-x86_64/egg/Crypto/SelfTest/Random/test_random.py:107: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 3:
build/bdist.linux-x86_64/egg/Crypto/Random/Fortuna/FortunaGenerator.py:28: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 2 and  sys.version_info[1] is 1:
build/bdist.linux-x86_64/egg/Crypto/Random/Fortuna/FortunaGenerator.py:28: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 2 and  sys.version_info[1] is 1:
zip_safe flag not set; analyzing archive contents...
Crypto.Cipher.__pycache__._AES.cpython-39: module references __file__
Crypto.Cipher.__pycache__._ARC2.cpython-39: module references __file__
Crypto.Cipher.__pycache__._ARC4.cpython-39: module references __file__
Crypto.Cipher.__pycache__._Blowfish.cpython-39: module references __file__
Crypto.Cipher.__pycache__._CAST.cpython-39: module references __file__
Crypto.Cipher.__pycache__._DES.cpython-39: module references __file__
Crypto.Cipher.__pycache__._DES3.cpython-39: module references __file__
Crypto.Cipher.__pycache__._XOR.cpython-39: module references __file__
Crypto.Hash.__pycache__._MD2.cpython-39: module references __file__
Crypto.Hash.__pycache__._MD4.cpython-39: module references __file__
Crypto.Hash.__pycache__._RIPEMD160.cpython-39: module references __file__
Crypto.Hash.__pycache__._SHA224.cpython-39: module references __file__
Crypto.Hash.__pycache__._SHA256.cpython-39: module references __file__
Crypto.Hash.__pycache__._SHA384.cpython-39: module references __file__
Crypto.Hash.__pycache__._SHA512.cpython-39: module references __file__
Crypto.PublicKey.__pycache__._fastmath.cpython-39: module references __file__
Crypto.Util.__pycache__._counter.cpython-39: module references __file__
Crypto.Util.__pycache__.strxor.cpython-39: module references __file__
creating /usr/local/lib/python3.9/dist-packages/pycrypto-2.6.1-py3.9-linux-x86_64.egg
Extracting pycrypto-2.6.1-py3.9-linux-x86_64.egg to /usr/local/lib/python3.9/dist-packages
/usr/local/lib/python3.9/dist-packages/pycrypto-2.6.1-py3.9-linux-x86_64.egg/Crypto/Random/Fortuna/FortunaGenerator.py:28: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 2 and  sys.version_info[1] is 1:
/usr/local/lib/python3.9/dist-packages/pycrypto-2.6.1-py3.9-linux-x86_64.egg/Crypto/Random/Fortuna/FortunaGenerator.py:28: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 2 and  sys.version_info[1] is 1:
/usr/local/lib/python3.9/dist-packages/pycrypto-2.6.1-py3.9-linux-x86_64.egg/Crypto/SelfTest/Random/test_random.py:107: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sys.version_info[0] is 3:
Adding pycrypto 2.6.1 to easy-install.pth file

Installed /usr/local/lib/python3.9/dist-packages/pycrypto-2.6.1-py3.9-linux-x86_64.egg
Searching for scapy==2.4.4
Best match: scapy 2.4.4
Adding scapy 2.4.4 to easy-install.pth file
Installing scapy script to /usr/local/bin

Using /usr/lib/python3/dist-packages
Searching for pyusb==1.0.2
Best match: pyusb 1.0.2
Adding pyusb 1.0.2 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for pyserial==3.5b0
Best match: pyserial 3.5b0
Adding pyserial 3.5b0 to easy-install.pth file
Installing pyserial-miniterm script to /usr/local/bin
Installing pyserial-ports script to /usr/local/bin

Using /usr/lib/python3/dist-packages
Finished processing dependencies for killerbee==3.0.0b0
                                                                                                                                                                                                               
┌──(kali㉿kali)-[~/killerbee]
└─$ zbid
           Dev Product String       Serial Number
[... 2 lines cut ...]
└─$ zbid
           Dev Product String       Serial Number
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b0', 'zbid')
  File "/usr/local/lib/python3.9/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python3.9/dist-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b0-py3.9-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 23, in <module>
    show_dev(gps=arg_gpsdev, include=args.include)
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b0-py3.9-linux-x86_64.egg/killerbee/__init__.py", line 46, in show_dev
    for dev in kbutils.devlist(vendor=vendor, product=product, gps=gps, include=include):
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b0-py3.9-linux-x86_64.egg/killerbee/kbutils.py", line 289, in devlist
    devlist = devlist_usb_v1x(vendor, product)
[... cut ...]
  File "/usr/lib/python3/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
                                                                                                                                                                                                               
┌──(kali㉿kali)-[~/killerbee]
└─$                                      
will I need to re-flash my AVR with the latest killerbee version?
probabluy

lucianoferrari, 2021-06-14:
I'll try it later

taylorcenters, 2021-06-22:
Wanted to check in on this. How's it working?

lucianoferrari, 2021-06-22:
> Wanted to check in on this. How's it working?

It's not working yet, but I was not able to update my RZ Raven USB flash with the latest version of Killerbee yet. I don't have the proper connector to plug in to the existing JTAG there. I'm trying to order an adaptor. If you have any suggestions I would appreciate. Thanks

taylorcenters, 2021-06-22:
I don't have that device, I've been using the Apimote - sorry

lucianoferrari, 2021-06-22:
> I don't have that device, I've been using the Apimote - sorry

Ohh, ok, where can I buy this device?

taylorcenters, 2021-06-22:
https://www.attify-store.com/products/apimote

lucianoferrari, 2021-06-22:
> https://www.attify-store.com/products/apimote

thank you! just bought one! :-)

SEC556, 2021-06-23:
FWIW, this Issue with zbid and the resultant "langid" errors also happens with the CC2531 with the Bumblebee 2.0 firmware:

Abbreviated lsusb output:

```
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 011: ID 0451:16a8 Texas Instruments, Inc. 
```
Output from zbid:

```
zbid
           Dev Product String       Serial Number
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b0', 'zbid')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1438, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.6/dist-packages/killerbee-3.0.0b0-py3.6-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 23, in <module>
    show_dev(gps=arg_gpsdev, include=args.include)
  File "/usr/local/lib/python3.6/dist-packages/killerbee-3.0.0b0-py3.6-linux-x86_64.egg/killerbee/__init__.py", line 46, in show_dev
    for dev in kbutils.devlist(vendor=vendor, product=product, gps=gps, include=include):
  File "/usr/local/lib/python3.6/dist-packages/killerbee-3.0.0b0-py3.6-linux-x86_64.egg/killerbee/kbutils.py", line 289, in devlist
    devlist = devlist_usb_v1x(vendor, product)
  File "/usr/local/lib/python3.6/dist-packages/killerbee-3.0.0b0-py3.6-linux-x86_64.egg/killerbee/kbutils.py", line 219, in devlist_usb_v1x
    usb.util.get_string(dev, dev.iProduct),     \
  File "/usr/lib/python3/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```

lucianoferrari, 2021-07-29:
OK, so I got my apimote and now I'm having different errors... When I try zbid it seems ok:
                                                                                                                                              
┌──(kali㉿kali)-[~]
└─$ sudo zbid              
           Dev Product String       Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2            
                
now when I try the zbstumbler I'm having some error messages:

┌──(kali㉿kali)-[~]
└─$ sudo zbstumbler -v
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
ERROR: Unable to inject packet: string argument without an encoding                                                                                                                                              
┌──(kali㉿kali)-[~]

Any idea what could be causing this? Thank you

lucianoferrari, 2021-07-29:
Also, when I'm doing some capture with wireshark using zbwireshark I get packets, but when I interact with the device this is what I get on the console:
zbwireshark: listening on '/dev/ttyUSB0', channel 15, page 0 (2425.0 MHz), link-type DLT_IEEE802_15_4, capture size 127 bytes
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'
# DEBUG b'Clearing overflow'

taylorcenters, 2021-07-29:
Hey, can you double check that you are running the latest firmware on the apimote, and running the latest version of killerbee, we've done a few PRs in the past few weeks.

lucianoferrari, 2021-07-30:
> Hey, can you double check that you are running the latest firmware on the apimote, and running the latest version of killerbee, we've done a few PRs in the past few weeks.

Thank you... is this the latest version? 
sudo zbid     
           Dev Product String       Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2

lucianoferrari, 2021-07-30:
> > Hey, can you double check that you are running the latest firmware on the apimote, and running the latest version of killerbee, we've done a few PRs in the past few weeks.
> 
> Thank you... is this the latest version?
> sudo zbid
> Dev Product String Serial Number
> /dev/ttyUSB0 GoodFET Api-Mote v2

Interesting also is that if I go to the killerbee directory and try to flash it I got an issue:
┌──(kali㉿kali)-[~/killerbee/firmware]
└─$ sudo ./flash_apimote.sh                                                                                                                                                                                1 ⨯
Flashing apimotev4_gf.hex to the connected USB serial device.
Traceback (most recent call last):
  File "./goodfet.bsl", line 34, in <module>
    import serial, os, subprocess, glob
ImportError: No module named serial

lucianoferrari, 2021-07-30:
Seems I'm using this version of killerbee - **KillerBee 3.0.0-beta Update**
and on api-mote.. I'm not quite sure how to check the version there...
sudo zbid
Dev Product String Serial Number
/dev/ttyUSB0 GoodFET Api-Mote v2

lucianoferrari, 2021-08-02:
> Hey, can you double check that you are running the latest firmware on the apimote, and running the latest version of killerbee, we've done a few PRs in the past few weeks.

I had tried to follow the same thing on Ubuntu and it worked better. I can use zbstumbler now. The only issue now is it continues to give me the DEBUG b'clearing overflow' error when I'm interacting with the device.

rmspeers, 2021-08-02:
For the ApiMote, to ensure you are at the latest version you can follow the flashing guidance in firmware/ folder.

For clearing overflow, that new firmware may help but it's also expected to have overflow when there is a lot of network traffic as if you can't empty it's FIFO buffer fast enough, you will have overflow.

Could you try to reduce the noise in your RF environment if overflow clearing is resulting in missed packets of interest? Or you could look to add a ring buffer in the MSP430 on the ApiMote to buffer things better in hardware than is done today.

lucianoferrari, 2021-08-03:
Seems I'm getting better results now! But I'm still having issues with the zbassocflood....
lferrari@ubuntu:~/killerbee$ sudo zbassocflood -p 0x1877 -c 11 -s .1
Auto-detection is being deprecated - Please specify hardware
zbassocflood: Transmitting and receiving on interface '/dev/ttyUSB0'
Traceback (most recent call last):
File "/usr/local/bin/zbassocflood", line 4, in
import('pkg_resources').run_script('killerbee==3.0.0b2', 'zbassocflood')
File "/usr/lib/python3/dist-packages/pkg_resources/init.py", line 667, in run_script
self.require(requires)[0].run_script(script_name, ns)
File "/usr/lib/python3/dist-packages/pkg_resources/init.py", line 1463, in run_script
exec(code, namespace, namespace)
File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbassocflood", line 153, in
assocreqinj = ''.join(assocreqp)
TypeError: sequence item 2: expected str instance, bytes found

bigthreede, 2021-08-30:
Hi,
i solved the installation problems with using the hints above for pyusb and libcrypt and was able to install killerbee on a kali 2021.2. But calling zbid gives me the following error:

─# sudo zbid                                                               1 ⨯
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbid')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 13, in <module>
    from killerbee import show_dev
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/killerbee/__init__.py", line 21, in <module>
    from .dot154decode import * #would like to import only within killerbee class
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/killerbee/dot154decode.py", line 5, in <module>
    from .exception import *
ModuleNotFoundError: No module named 'killerbee.exception'

Any ideas what could be wrong and how i could fix that?

taylorcenters, 2021-09-02:
sorry, I think in killerbee/__init__.py comment out the import for .dot154decode should resolve. I'll keep looking and make sure that doesn't break more things though.

bigthreede, 2021-09-03:
Hi, commenting out that import seems to have no effects, i get same error messages as before...

nourmehdi, 2021-09-05:
Hi , got the same issue , when trying to run zbstumbler on ubuntu20.04 ,
 Last error message , File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/dot154decode.py", line 5, in <module>
    from .exception import *
ModuleNotFoundError: No module named 'killerbee.exception'   
after some digging , fount that the import .exception in dot154decode.py , is reffering to a local folder or module wich doesn't exist in the killerbee folder . After commenting out the import .dot154decode.py in the init file , things seems to work normaly.

bigthreede, 2021-09-06:
ok, with a reinstallation and commenting out the import of .dot154decode.py i got at least zbid to run. But i have problems now with zbfind, it needs PyGtk 2.3.93 or later, but PyGtk is not available for Kali2021.2. Do you have a solution for that too?

taylorcenters, 2022-03-08:
I'm sorry I can't do anything for PyGtk on Kali, you could take the zbfind script and remove the gtk parts and send output to files or command line.

this branch removed that non-existant import, and addresses some of the cc2531 bugs mentioned earlier in this ticket. https://github.com/riverloopsec/killerbee/tree/bugfix/cc2531-assorted.

Made a change to that line in zbassoc flood to join the array into a bytestring.

I appreciate everyone's input and help on these issues. This topic has wandered so I'm going to close. If you still have issues and need help please open another issue for the specific problem!

closed 2022-03-08 by taylorcenters