riverloopsec/killerbee #234: TypeError: 'type' object is not subscriptable

blackndoor, 2021-07-13:
I followed the installation:
```
$ git clone https://github.com/riverloopsec/killerbee.git && cd killerbee
$ sudo pip3 install pyusb
$ sudo python3 setup.py install
[...]
Finished processing dependencies for killerbee==3.0.0b1
```

If I try to find my apimote:

```
/home/bkndr# zbid
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b1', 'zbid')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1446, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.7/dist-packages/killerbee-3.0.0b1-py3.7-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 13, in <module>
    from killerbee import show_dev
  File "/usr/local/lib/python3.7/dist-packages/killerbee-3.0.0b1-py3.7-linux-x86_64.egg/killerbee/__init__.py", line 8, in <module>
    from .daintree import *
  File "/usr/local/lib/python3.7/dist-packages/killerbee-3.0.0b1-py3.7-linux-x86_64.egg/killerbee/daintree.py", line 69, in <module>
    class DainTreeReader:
  File "/usr/local/lib/python3.7/dist-packages/killerbee-3.0.0b1-py3.7-linux-x86_64.egg/killerbee/daintree.py", line 94, in DainTreeReader
    def pnext(self) -> Optional[list[Any]]:
TypeError: 'type' object is not subscriptable

```

blackndoor, 2021-07-13:
Well, it works fine with the branch py3 :), but:
```
$ sudo zbid         
           Dev Product String       Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2

$ sudo zbstumbler -v
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
ERROR: Unable to inject packet: string argument without an encoding
```

taylorcenters, 2021-07-13:
I'll take a look at this today, thanks for reporting! What platform are you running this on? Ubuntu?

taylorcenters, 2021-07-13:
Hey check out this branch zbid_type_error https://github.com/riverloopsec/killerbee/pull/235 and let me know if this has better results. I was able to reproduce what you saw and this resolved it for me.

taylorcenters, 2021-07-13:
well.. I fixed the type stuff, let me see if I can figure out the zbstumbler issue. Note to self: write tests for kb tools.

taylorcenters, 2021-07-13:
zbstumbler issue should be resolved. I intend to clean up the tools more in a future PR. This one should immediately address these bugs.

taylorcenters, 2021-07-14:
This change has been merged into the develop branch. Please update if this has fixed your issue. Thanks!

blackndoor, 2021-07-15:
Thank you! I installed the develop branch.
zbid:
```
~/tools/killerbee(develop) $ sudo zbid
[sudo] Mot de passe de bkndr : 
           Dev Product String                 Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2
```
zbstumbler:
```
~/tools/killerbee(develop) $ sudo zbstumbler -v
Auto-detection is being deprecated - Please specify hardware
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
[stuck?]
```

taylorcenters, 2021-07-16:
looks like we're not actually doing the timeout in the apimote pnext. Sorry about that. I'll go work on that now. Thanks for your help on this.

taylorcenters, 2021-07-16:
try new branch apimote-timeout, will default to 2s delay between channels, you can use the -s, --delay flag to override that with an int number of seconds to wait on each channel.

blackndoor, 2021-07-19:
here is the output:

```
~/tools/killerbee(apimote-timeout) $ sudo zbstumbler -v
Auto-detection is being deprecated - Please specify hardware
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
Setting channel to 12.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
^C
1 packets transmitted, 0 responses.

~/tools/killerbee(apimote-timeout) $ sudo zbstumbler -v -s 1
Auto-detection is being deprecated - Please specify hardware
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Setting channel to 11.
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
Setting channel to 12.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
^C
1 packets transmitted, 0 responses.
```

taylorcenters, 2021-07-19:
in the develop branch did it seem like the -s 1 delay flag was working? did the channel switch every 1 sec? Before it looked like you were hanging (and I experienced that too) I wonder why it appears to work correctly for you now.

@rmspeers any ideas, I don't see where the changes in the apimote-timeout branch would cause the serial timeouts that are absent in the develop branch.

blackndoor, 2021-07-19:
Yes, with the develop branch the -s 1 delay flag works (the channel switches every 1 sec). With the apimote-timeout branch and the delay flag (1s), the channel switch (only once) after 10 sec.

taylorcenters, 2021-07-19:
and just to double check, you are running `python setup.py install` when you switch branches?

blackndoor, 2021-07-19:
yes after each git clone.

lucianoferrari, 2021-08-03:
> yes after each git clone.
> 
> Edit: well I use `sudo python3 setup.py install`

that's what I do

lucianoferrari, 2021-08-03:
Seems I'm getting better results now! But I'm still having issues with the zbassocflood....
lferrari@ubuntu:~/killerbee$ sudo zbassocflood -p 0x1877 -c 11 -s .1
Auto-detection is being deprecated - Please specify hardware
zbassocflood: Transmitting and receiving on interface '/dev/ttyUSB0'
Traceback (most recent call last):
  File "/usr/local/bin/zbassocflood", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbassocflood')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbassocflood", line 153, in <module>
    assocreqinj = ''.join(assocreqp)
TypeError: sequence item 2: expected str instance, bytes found

taylorcenters, 2022-03-08:
please try the newest version of kb, I'm having success with zbstumbler - On branch https://github.com/riverloopsec/killerbee/tree/bugfix/cc2531-assorted

% zbstumbler -i /dev/tty.usbserial-4CA07TW9
zbstumbler: Transmitting and receiving on interface '/dev/tty.usbserial-4CA07TW9'
# DEBUG b'Clearing overflow'
New Network: PANID 0x91F3 Source 0xA41B
Ext PANID: 00:00:00:00:02:00:00:4F Stack Profile: Network Specific
Stack Version: ZigBee Prototype
Channel: 15

taylorcenters, 2022-06-29:
Hopefully this PR resolved your issue. Feel free to re-open if this is still relevant. Thanks!

closed 2022-06-29 by taylorcenters