riverloopsec/killerbee #244: Zbstumbler error when finding the right channel

xsoltek, 2021-11-04:
Hello here, i have a correct detection with zbid, i had already reflash my apimote with the 2 hex files in the firmware folder. I have uninstall/ reinstall killerbee but nothing change. im using Ubuntu 20.04.

Moreover, Zbwireshark is working but displays BAD FCS and in the console # DEBUG b 'Clearing overflow' 

do you guys have any ideas ? 

Thanks

taylorcenters, 2022-03-08:
Hi, unfortunately these kind of errors are expected with the Apimote as the FIFO buffer overflows, especially in noisy environments. I have had much better experience using a CC2531 from TI for sniffing, and I use the Apimote for injecting/sending traffic.

taylorcenters, 2022-03-08:
maybe try the newest version of kb, I'm having success with zbstumbler

% zbstumbler -i /dev/tty.usbserial-4CA07TW9
zbstumbler: Transmitting and receiving on interface '/dev/tty.usbserial-4CA07TW9'
# DEBUG b'Clearing overflow'
New Network: PANID 0x91F3 Source 0xA41B
        Ext PANID: 00:00:00:00:02:00:00:4F      Stack Profile: Network Specific
        Stack Version: ZigBee Prototype
        Channel: 15

taylorcenters, 2022-06-29:
Hopefully this PR resolved your issue. Feel free to re-open if this is still relevant. Thanks!