riverloopsec/killerbee #245: no module named 'killerbee.exception'

lionelmcpint, 2021-11-04:
Hi guys,

I'm trying to install KillerBee to use as part of a project for university and when I try to run any of the commands it throws the exact same error for each command:

```
┌──(lionel㉿kali)-[~/Desktop]
└─$ zbid                                                                                           127 ⨯
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbid')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 13, in <module>
    from killerbee import show_dev
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/killerbee/__init__.py", line 21, in <module>
    from .dot154decode import * #would like to import only within killerbee class
  File "/usr/local/lib/python3.9/dist-packages/killerbee-3.0.0b2-py3.9-linux-x86_64.egg/killerbee/dot154decode.py", line 5, in <module>
    from .exception import *
ModuleNotFoundError: No module named 'killerbee.exception'
```

I've run every command available and they are all listing exactly the same issue. Does anyone have any suggestions as to why this is happening? I plan on trying to review some older versions to see if there was an issue upgrading, however software development is not my strong point - this is for a security project.

Any help would be greatly appreciated as I have built the foundation of my honours project around this tool working, and I do not want to have to try and change so late in the game.

Many thanks,
Lionel

xsoltek, 2021-11-05:
Hi lionelmcpint, you just have to comment out in the __init__.py the line "from .dot154decode import * #would like to import only within killerbee class" and u will be able to start zbid or other tools. 

Happy hacking :)

taylorcenters, 2021-11-05:
Hi, Thanks xsoltek! I believe that change has been made in the develop branch, but hasn't been packaged into a tagged version yet.

Lionel, please let me know if that change works, and if you run into other issues. I'm trying to make some time this month to address a lot of the other bugs people are running into this month. I want to make sure you're able to use this for your project.

lionelmcpint, 2021-11-09:
Hi guys,

Thanks very much for the quick reply, commenting out that option in the init.py solved the issue! I've spent the weekend testing and it everything else seems to work.

Thanks again for all the hard work maintaining, I'm looking forward to using the framework properly!

All The Best, 
Lionel

closed 2021-11-09 by lionelmcpint