riverloopsec/killerbee #247: CC2531 and Killerbee - current version stable?

Atzeloss, 2021-11-25:
Hi,
I have tried to use the killerbee framework with my CC2531 stick together with the 'Bumblebee' firmware. 
My used platform was a freshly installed Ubuntu 20.04.3 LTS virtual machine.
Bumblebee firmware: https://github.com/virtualabs/cc2531-killerbee-fw

Also, I didn't run into any errors during the installation after changing the library installation commands from the readme file to use Python 3, and commenting out the problematic line mentioned in #245.
However, it seems to me that the sniffing killerbee commands do not work as intended.
*Note: the readme file mentions that only sniffing functionality is supported on the CC2351.*

Therefore, I have expected three commands to be compatible with the CC2351 stick.
- zbid (discovery of devices)
- zbdump (sniffing and dumping traffic in pcap file)
- zbwireshark (sniffing and piping traffic into wireshark)

However, none of them worked as intended. 
Instead, I have encountered various errors:

## zbid:
```Wireshark exited (0)
0 packets captured
basti@ubuntu:~/Desktop/killerbee$ zbid
           Dev Product String                 Serial Number
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbid')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 23, in <module>
    show_dev(gps=arg_gpsdev, include=args.include)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/__init__.py", line 36, in show_dev
    for dev in devlist(vendor=vendor, product=product, gps=gps, include=include): 
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/kbutils.py", line 271, in devlist
    devlist: List[Any] = devlist_usb_v1x(vendor, product)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/kbutils.py", line 221, in devlist_usb_v1x
    usb.util.get_string(dev, dev.iProduct),     \
  File "/usr/lib/python3/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```

## zbdump:
This seems to be a problem with the python-usb library interface. As mentioned in the dev_cc253x.py file, the python-usb 1.x library uses a usb.core.Device object, instead of a string. 
Also, I have tried to fix this problem myself, but I wasn't able to.

```Warning: You are using pyUSB 1.x, support is in beta.
Traceback (most recent call last):
  File "/usr/local/bin/zbdump", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbdump')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbdump", line 159, in <module>
    main()
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbdump", line 135, in main
    kb = KillerBee(device=args.devstring, hardware=args.device)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/__init__.py", line 93, in __init__
    self.driver = CC253x(device, self.__bus, CC253x.VARIANT_CC2531)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/dev_cc253x.py", line 61, in __init__
    self.dev.set_configuration()
AttributeError: 'str' object has no attribute 'set_configuration'
```

##zbwireshark 
- Wireshark opens without problems but no packets are recorded/displayed
- However: The CC2531 stick "lights up" every time, I am sending a packet - Therefore I think the packets are somehow registered

My question now is whether my problems are coming from a faulty installation, or whether the problems are being caused by the Killerbee framework (considering the CC2531 usb device)?

yrjo, 2022-01-29:
Hi, would like to +1 as well :)

Menzoc, 2022-02-07:
Hi!
I'm strunggling myself with this problem.
My USB dongle give me some answer:

## zbid
Did you launch your command in sudo?
## zbdump
Can you check the '-d' option I think you need to enter 'bumblebe'
```
sudo zbdump -i 1:29 -f 11 -w out.dump -d bumblebe -n 100
```
## zbwireshark
And here is the part where I'm getting blocked a lot of time...
With bumblebee, you cannot sniff any packet (with any command) and I don't have any clue why.
In the end, I changed my version of killerbee I go to: https://github.com/virtualabs/killerbee/
I know that is not a good way to fix the issue but I was not able to find the problem...

taylorcenters, 2022-06-29:
Hopefully this PR resolved your issue. Feel free to re-open if this is still relevant. Thanks!

closed 2022-06-29 by taylorcenters

h4knet, 2022-10-12:
Hello,
I had the same issues as you guys. The firmware in v2.1 is only supported with the fork if Virtualabs but the fork misses some commits with fixes.

I was able to receive packet in wireshark by using the curent latest version of killerbee (develop branch) and flashing the firmware in version 2.0. 

I tried v1.1 as @taylorcenters suggested but it didn't worked for me.