riverloopsec/killerbee #249: CC253x - zbwireshark/zbdump - TypeError: a bytes-like object is required, not 'int'

Menzoc, 2022-02-03:
Hello there!

I'm trying to install and test kilerbee. I just finished the installation procedure. 
zbid is working:
```
sudo zbid 
           Dev Product String                 Serial Number
          1:14 CC2531 USB Dongle                    None
```
However, I've issues to understand how to use zbwireshark and zbdump. I try the following request:
```
sudo zbwireshark -i 1:14 -c 11 
sudo zbdump -i 1:14 -d CC253x -f 11 -w first_file.pcap
```
I received the same error: 
```
Warning: You are using pyUSB 1.x, support is in beta.
zbwireshark: listening on 'CC2531 USB Dongle', channel 11, page 0 (2405.0 MHz), link-type DLT_IEEE802_15_4, capture size 127 bytes
Traceback (most recent call last):
  File "/usr/local/bin/zbwireshark", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbwireshark')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 667, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1463, in run_script
    exec(code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 108, in <module>
    main()
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/EGG-INFO/scripts/zbwireshark", line 84, in main
    packet = kb.pnext()
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/__init__.py", line 442, in pnext
    return self.driver.pnext(timeout)
  File "/usr/local/lib/python3.8/dist-packages/killerbee-3.0.0b2-py3.8-linux-x86_64.egg/killerbee/dev_cc253x.py", line 250, in pnext
    rssi = struct.unpack("b", framedata[-2])[0] - 73
TypeError: a bytes-like object is required, not 'int'
```
As I see that framedata is a bytes-likes object, I didn't understand why do we have a type error...

Can someone explain me what I've missed please? Is it an installation mistake?

rmspeers, 2022-02-03:
As such, try changing:
```
    rssi = struct.unpack("b", framedata[-2])[0] - 73
```
to
```
    rssi = framedata[-2] - 73
```

See if that works?

Menzoc, 2022-02-03:
It's Worling! Thank you for your fast answer and help!

closed 2022-02-03 by Menzoc