riverloopsec/killerbee #250: Cannot inject packets on RZ USB stick with Python 3 (TypeError)

nitrescov, 2022-02-10:
Hey all,
I want to share a small issue I've noticed when using a RZ USB stick with killerbee.

## General information

- OS: Kali Linux 2022.1
- Python 3.9.9
- Killerbee version 3.0.0-beta.2
- Device: AVR RZ USB stick (with killerbee firmware)

## Issue

With Python 2, scapy packets could be passed as str(pkt) to the killerbee inject method, which worked fine.

With Python 3, on the other hand, it doesn't matter whether you pass the package as a string or bytes. Killerbee always tries to concatenate both data types.

### Traceback 1: injecting bytes

```python
$ python3 test_inject.py
Traceback (most recent call last):
  File ".../test_inject.py", line 41, in <module>
    kb.inject(bytes(Dot15d4()))
  File ".../killerbee/dev_rzusbstick.py", line 523, in inject
    packet += "\x00\x00"
TypeError: can't concat str to bytes
```

Here the problem is clear.

### Traceback 2: injecting string

```python
$ python3 test_inject.py
Traceback (most recent call last):
  File ".../test_inject.py", line 41, in <module>
    kb.inject("".join(map(chr, bytes(Dot15d4()))))  # replacement for str(pkt) in Python 2
  File ".../killerbee/dev_rzusbstick.py", line 528, in inject
    self.__usb_write(RZ_USB_COMMAND_EP, struct.pack("BB", RZ_CMD_INJECT_FRAME, len(packet)) + packet)
TypeError: can't concat str to bytes
```

In this case it seems to be the other way round. Apparently struct.pack() returns bytes which cannot be concatenated with the packet string.

This can also be confirmed with killerbee internal tools that use the inject function (e.g. zbstumbler).

Has anyone else noticed this problem so far?

taylorcenters, 2022-03-14:
Unfortunately, I don't have an rzusbstick to specifically test this, but after mocking the function to confirm the change you suggested I have added it to PR #252 - Thanks for finding and solving this one!

nitrescov, 2022-03-15:
Thank you for adding the fix. I made several tests with the RZUSB stick and can confirm that sending packets will work again with this simple change. I just didn't have time to make a pull request yet ;)

taylorcenters, 2022-06-29:
Thank you again for your help!

closed 2022-06-29 by taylorcenters