riverloopsec/killerbee #251: zbfind tool not working

lionelmcpint, 2022-02-24:
Hi again guys,

I'm having trouble getting the zbfind tool working to display the RSSI values of received signals.
At the moment I'm using the following setup:
**OS**: Ubuntu 18.04
**Zigbee Version**: 2.7.2
**Hardware**: TI CC2531
**Python**: 2.7.17

The error message recieved is as follows:
```
root@killerbee-18-04-03:~# zbfind -c 25
Warning: You are using pyUSB 1.x, support is in beta.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/killerbee-2.7.2-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbfind", line 1039, in run
AttributeError: type object 'datetime.datetime' has no attribute 'datetime'
```

I am aware that this issue has been raised previously, however the update was seemingly changing the code to exactly what it is now.

There are no issues running _zbdump_ or _zbwireshark_ (I am currently waiting on delivery of my Sewio OpenSniffer to test the injection tools), I am only having issues with the _zbfind_ tool.
My reasons for using an older version of zigbee instead of the development branch is that I encounter nothing but errors when trying to run anything in v3.0.0, regardless of which OS or config I try.
My research requires me to record the RSSI measurements of incoming signals, and as far as I can tell the zbfind tool is the only way to record this data. 

If anyone can help me either fix zbfind or offer an alternative method of obtaining RSSI this would be greatly appreciated. I have tried as hard as I can to resolve this issue however it is seemingly outwith my skillset.

Thank you,
Lionel

taylorcenters, 2022-03-08:
Hi please try using the python3 version of the package - this branch addresses a lot of the current issues with cc2531 https://github.com/riverloopsec/killerbee/tree/bugfix/cc2531-assorted

taylorcenters, 2022-06-29:
Hopefully this PR resolved your issue. Feel free to re-open if this is still relevant. Thanks!

closed 2022-06-29 by taylorcenters