riverloopsec/killerbee #253: transport key arg in zbdsniff

jamesleytevidal, 2022-04-14:
Hi there, 

Working with a newer version of zbdsniff than I'm used to (for reference, the versions I've used previously supported DCF, so a while back...) 

I don't see any documentation on the transport key (-k) argument. I have pcaps with a transport key in them but zbdsniff refuses to extract them without the -k value...

apologies for not connecting the dots myself - if I have a pcap with a key in it, what value should I be using for -k to extract said key?

taylorcenters, 2022-04-22:
Haven't had a chance to look too much into this. Would the key it's expecting be ZigBeeAlliance09?

jamesleytevidal, 2022-04-22:
So, out of curiosity I did try that, presuming it was in hex I used 5a6967426565416c6c69616e63653039 and got this. 

[image]

However, I'm seeing this error regardless of key or format of the key that I've tried so far.

jamesleytevidal, 2022-07-22:
My concern is resolved with https://github.com/riverloopsec/killerbee/pull/260 if that can make it's way back into your codebase.