riverloopsec/killerbee #254: error when running setup.py

Damien455, 2022-04-21:
building 'zigbee_crypt' extension
x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/local/include -I/usr/include -I/sw/include/ -Izigbee_crypt -I/usr/include/python3.8 -c zigbee_crypt/zigbee_crypt.c -o build/temp.linux-x86_64-3.8/zigbee_crypt/zigbee_crypt.o
x86_64-linux-gnu-gcc: error: zigbee_crypt/zigbee_crypt.c: No such file or directory
x86_64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

taylorcenters, 2022-04-22:
are you able to see the file zigbee_crypt/zigbee_crypt.c ?

Eofric, 2022-05-05:
I had the same issue.  Installing libgcrypt-dev via apt solved it.

freqyXin, 2023-03-08:
Still having this issue on MacOS.

Libgcrypt-dev is not available via homebrew or macports.

```
building 'zigbee_crypt' extension
clang -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch arm64 -arch x86_64 -g -I/usr/local/include -I/usr/include -I/sw/include/ -Izigbee_crypt -I/Library/Frameworks/Python.framework/Versions/3.11/include/python3.11 -c zigbee_crypt/zigbee_crypt.c -o build/temp.macosx-10.9-universal2-cpython-311/zigbee_crypt/zigbee_crypt.o
zigbee_crypt/zigbee_crypt.c:15:10: fatal error: 'gcrypt.h' file not found
#include <gcrypt.h>
         ^~~~~~~~~~
1 error generated.
error: command '/usr/bin/clang' failed with exit code 1 

```

Any advice?

mma4194, 2023-11-15:
Had the same issue and solved it using: `sudo apt-get install -y libgcrypt-dev`