riverloopsec/killerbee #255: syntax error after flashing apimote

bazard89, 2022-05-03:
Flashed apimote twice as others have. it completed the second time but now when ever I use any zb command I get the following

r@server-virtual-machine:~/Attify-Zigbee-Framework-master/firmware$ zbid
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbid')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1469, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/killerbee-3.0.0b2-py2.7-linux-x86_64.egg/EGG-INFO/scripts/zbid", line 13, in <module>
 
  File "/usr/local/lib/python2.7/dist-packages/killerbee-3.0.0b2-py2.7-linux-x86_64.egg/killerbee/__init__.py", line 25
    def show_dev(vendor: str=None, product: str=None, gps: str=None, include: str=None) -> None:
                       ^
SyntaxError: invalid syntax
 
was previously getting "debug b'clearing overflow" for all packets picked up by zbstumbler as well.

bazard89, 2022-05-03:
well, that got sorted out after reflashing to app to apimotev4_202106.hex and reinstalling killerbee, but now when running zstumbler, instead of  "debug b'clearing overflow" it now it looks like this - 

r@server-virtual-machine:~/Attify-Zigbee-Framework-master/firmware$ sudo zbstumbler 
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning, failed to set r11=0x0020, got 00.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).

taylorcenters, 2022-05-03:
I noticed in the first output you were using python2.7 ... upon reinstall of killerbee are you using python3 ?

bazard89, 2022-05-20:
just did and issue persist. if i put the apimote_gf.hex firemware i just get the overflow message. i cannot get any info other than dev/ttyUSB0 when doing zbid ID either, which I suspect is the cause

taylorcenters, 2022-05-20:
can you show output to zbid and zbstumbler with new python version please

ArcticNarwhal, 2022-11-05:
I'm having the same issue, flashed  ApiMote v4b board with `board=apimote3 python2 ./goodfet.bsl -e -p 'apimotev4_202106.hex'`

```
[hugo@hugos-laptop ~]$ zbid
           Dev Product String                 Serial Number
  /dev/ttyUSB0 GoodFET Api-Mote v2                      
[hugo@hugos-laptop ~]$ python --version
Python 3.10.7
[hugo@hugos-laptop ~]$ pip3 list
Package            Version
------------------ ---------
argcomplete        2.0.0
Beaker             1.10.0
beautifulsoup4     4.11.0
blivet             3.4.4
blivet-gui         2.3.0
Brlapi             0.8.4
cffi               1.15.0
chardet            4.0.0
charset-normalizer 2.0.11
click              8.0.4
cryptography       36.0.0
cupshelpers        1.0
dasbus             1.6
dbus-python        1.2.18
distro             1.6.0
fedora-third-party 0.10
fros               1.1
gpg                1.17.0
gTTS               2.2.2
gTTS-token         1.1.4
humanize           3.13.1
idna               3.3
killerbee          3.0.0b2
langtable          0.0.60
libcomps           0.1.18
lxml               4.7.1
Mako               1.1.4
MarkupSafe         2.1.1
nftables           0.1
numpy              1.22.0
olefile            0.46
packaging          21.3
Paste              3.5.0
pexpect            4.8.0
pid                2.2.3
Pillow             9.1.0
pip                21.3.1
ply                3.11
productmd          1.33
ptyprocess         0.6.0
pwquality          1.4.4
pycairo            1.21.0
pycparser          2.20
pycrypto           2.6.1
pycups             2.0.1
pycurl             7.45.1
pyenchant          3.2.2
PyGObject          3.42.1
pykickstart        3.36
pyOpenSSL          21.0.0
pyparsing          2.4.7
pyparted           3.12.0
pyserial           3.5
PySocks            1.7.1
python-augeas      1.1.0
python-dateutil    2.8.1
python-meh         0.50
pyudev             0.22.0
pyusb              1.2.1
pyxdg              0.27
RangeParser        0.1.3
regex              2022.9.13
requests           2.27.1
requests-file      1.5.1
requests-ftp       0.3.1
rpm                4.17.1
scapy              2.4.5
selinux            3.3
sepolicy           3.3
setools            4.4.0
setuptools         59.6.0
simpleaudio        1.0.4
simpleline         1.9.0
six                1.16.0
sos                4.3
soupsieve          2.3.1
systemd-python     234
Tempita            0.5.2
urllib3            1.26.12
[hugo@hugos-laptop ~]$ zbstumbler -c 15 -v
zbstumbler: Transmitting and receiving on interface '/dev/ttyUSB0'
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
Transmitting beacon request.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
Warning, failed to set r11=0xe0f2, got e0d2.
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
^C
2 packets transmitted, 0 responses.
[hugo@hugos-laptop ~]$ python
Python 3.10.7 (main, Sep  7 2022, 00:00:00) [GCC 12.2.1 20220819 (Red Hat 12.2.1-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import killerbee
>>> kb = killerbee.KillerBee()
>>> kb.get_dev_info()
['/dev/ttyUSB0', 'GoodFET Apimote v2', '']
>>> kb.sniffer_on(channel=15)
>>> kb.pnext()
Warning: waiting for serial read timed out (most likely).
{0: b'\xe0\xce', 1: False, 2: 78, 'bytes': b'\xe0\xce', 'validcrc': False, 'rssi': 78, 'location': None, 'datetime': datetime.datetime(2022, 11, 5, 14, 25, 58, 158959), 'dbm': 33}
>>> kb.pnext()
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
{0: b'', 1: False, 2: 128, 'bytes': b'', 'validcrc': False, 'rssi': 128, 'location': None, 'datetime': datetime.datetime(2022, 11, 5, 14, 26, 25, 411637), 'dbm': 83}
>>> kb.pnext()
Warning: waiting for serial read timed out (most likely).
Warning: waiting for serial read timed out (most likely).
{0: b'', 1: False, 2: 128, 'bytes': b'', 'validcrc': False, 'rssi': 128, 'location': None, 'datetime': datetime.datetime(2022, 11, 5, 14, 26, 54, 896526), 'dbm': 83}
>>> exit()
```

Going back to `apimotev4_gf.hex` returns functionality.  
When using  zbstumbler, one beacon request (with valid fcs) is sent.