riverloopsec/killerbee #256: i suspect there is something wrong about the mac footer calculated

jbgod, 2022-05-19:
i had use killerbee to write a tool for zigbee network connect.
with a threading capture process to sniffer,just use the killerbee to send zigbee packet.

i found something that  make me feel so weird.
i imitate as a normal zigbee device to send association request, but the zigbee gateway response with a Pan access denied
[image]
[image]

jbgod, 2022-05-19:
i had try apimote and cc2531

taylorcenters, 2022-05-19:
the apimote should be letting the radio auto generate the crc if you use the inject() function

https://github.com/riverloopsec/killerbee/blob/c4137618263ffdb324ff86ed8a7875c3282fc1f1/killerbee/dev_apimote.py#L182
        self.handle.RF_autocrc(1)               #let radio add the CRC

the RF_txpacket is sending a command to the apimote firmware - you can see the code to handle it here:
https://github.com/travisgoodspeed/goodfet/blob/master/firmware/apps/radios/ccspi.c
https://github.com/travisgoodspeed/goodfet/blob/master/firmware/include/ccspi.h

the CCSPIAPP tells it to use this app - 0x81 is the "verb" to use the ccspi_tx function - then has the packet bytes+length to send

jbgod, 2022-05-20:
def test_decrypt_cm(self):
        key = b'\x00\x01\x02\x03\x04\x05\x06\x07\x08\x09\x0a\x0b\x0c\x0d\x0e\x0f'
        nonce = b'\x00\x01\x02\x03\x04\x05\x06\x07\x08\x09\x0a\x0b\x0c'
        enc_data = b'\x17\x36\xb7\x8c'
        mic = b'\xfc\xe0\xce\x86'
        zigbee_data = b'\x00\x00\x00\x00'
     
        (pt_data, mic_check) = decrypt_ccm(key, nonce, mic, enc_data, zigbee_data)

i guess enc_data is aps_payload in packet with pkcs7padding, what is the args zigbee_data.
i capture the transport key packet, i want to decrypt it

taylorcenters, 2022-05-20:
can find implementation for decrypt_ccm here:
https://github.com/riverloopsec/killerbee/blob/develop/zigbee_crypt/zigbee_crypt.c

The parameters for this function are:
key: 16-byte decryption key
nonce: 13-byte nonce
mic: 4-16 byte message integrity check (MIC)
encrypted_payload: The encrypted data to decrypt
zigbee_data: The zigbee data within the frame, without the encrypted payload, MIC, or FCS

the zigbee_data is used as auth-data, so it will need to match the same value used during encryption. The decrypted data should be unaffected, the MIC validation will be false if this value doesn't match.

jbgod, 2022-05-20:
python3 test.py
32
b'\x94J\x84\xfe\xff\xbd\x1b\xec\t\xa0\x05\x000'
Traceback (most recent call last):
  File "test.py", line 103, in <module>
    zigbee_aps_decrypt(data)
  File "test.py", line 96, in zigbee_aps_decrypt
    result = kbdecrypt(scapy_packet, key)
  File "/usr/local/lib/python3.7/site-packages/killerbee/scapy_extensions.py", line 474, in kbdecrypt
    nonce: bytes = struct.pack('L',source_pkt[ZigbeeNWK].ext_src)+struct.pack('I',source_pkt[ZigbeeSecurityHeader].fc) + sec_ctrl_byte
struct.error: required argument is not an integer

my code

    scapy_packet = Dot15d4(data)
    result = kbdecrypt(scapy_packet, key)

i try the function in kbdecrypt,some error happen

taylorcenters, 2022-05-20:
hmm failing here: https://github.com/riverloopsec/killerbee/blob/3bbf7eb147ce9b94734ef17a1659e36690251d25/killerbee/scapy_extensions.py#L474

from your code I see

```
>>> scapy_packet = Dot15d4(data)
>>> scapy_packet[ZigbeeNWK].ext_src
>>> scapy_packet[ZigbeeSecurityHeader].fc
368649
```

so it's failing because your packet isn't using an ext_src. I'll keep looking to see how to fix.

jbgod, 2022-05-20:
i had try  decrypt_ccm. the aes-ccm* result is wrong 
the packet is transport key packet to send encrypted network key

the packet is(copy from ubiqua)

(encrypted packet data)
61 88 EC 21 38 B8 6F 00 00 08 00 B8 6F 00 00 1E 7E 21 C4 30 09 A0 05 00 94 4A 84 FE FF BD 1B EC 
(encrypted aps payload)
DD AE 2F 3E DE 19 46 66 B5 DE 48 75 48 EC 98 42 5D 20 12 FA B4 5D 38 3E 59 C8 37 A9 79 30 C6 70 CE 16 EF 
(APS MIC)
4D DF 2E B9 
FF FF

(decrypted packet data)
(APS payload)
01 47 F3 20 01 83 1C 1C B6 43 A1 45 7F 3F 80 D9 9D 00 88 31 CA FE FF F9 E3 B4 94 4A 84 FE FF BD 1B EC
(aps MIC)
49 18 A0 1E

`
from Crypto.Cipher import AES
from Crypto.Util import Counter
import binascii
from zigbee_crypt import *
from killerbee.scapy_extensions import *

data = b'a\x88\xec!8\xb8o\x00\x00\x08\x00\xb8o\x00\x00\x1e~!\xc40\t\xa0\x05\x00\x94J\x84\xfe\xff\xbd\x1b\xec\xdd\xae/>\xde\x19Ff\xb5\xdeHuH\xec\x98B] \x12\xfa\xb4]8>Y\xc87\xa9y0\xc6p\xce\x16\xefM\xdf.\xb97\xeb'
def zigbee_aps_decrypt(data): 
    #select mac payload         
    #nwk header  8 bytes | nwk payload
    mac_payload = data[9 : -2]
    nwk_header = mac_payload[:8]
    #select nwk payload        
    #aps header (2 bytes) | aps aux header (13 bytes) | aps payload 
    #nwk_payload = data[17: -2]
    nwk_payload = mac_payload[8:]
    aps_header = nwk_payload[:2]

    aps_aux_header = nwk_payload[2:15]
    # content in APS AUX Header
    # APS Security Control (1 bytes) | APS Frame Counter (4 bytes) | Source Address (8 bytes)
    #print(aps_aux_header)
    aps_security_control = aps_aux_header[0]
    aps_frame_counter = aps_aux_header[1:5]
    #print(aps_frame_counter)
    source_address = aps_aux_header[5:]
    #print(source_address)

    aps_payload = nwk_payload[15:-4]
    print(len(aps_payload))
    aps_mic = nwk_payload[-4:]
    #print(aps_payload)
    #add_auth_data
    auth_data = nwk_header + aps_aux_header
    add_auth_data = len(auth_data).to_bytes(2, byteorder = 'big') + auth_data
    add_auth_data = pkcs7padding(add_auth_data)
    #nonce
    nonce = source_address + aps_frame_counter + bytes([aps_security_control])
    #print(nonce)
    enc_data = aps_payload
    enc_data1 = pkcs7padding(aps_payload)
    zigbee_data = b'a\x88\xec!8\xb8o\x00\x00\x08\x00\xb8o\x00\x00\x1e~!\xc40\t\xa0\x05\x00\x94J\x84\xfe\xff\xbd\x1b\xec'
    key = bytes([0x5A, 0x69, 0x67, 0x42, 0x65, 0x65, 0x41, 0x6C, 0x6C,0x69,0x61,0x6E,0x63, 0x65, 0x30, 0x39])
    
    (pt_data, mic_check) = decrypt_ccm(key, nonce, aps_mic, enc_data, zigbee_data)
    (pt_data1, mic_check1) = decrypt_ccm(key, nonce, aps_mic, enc_data1, zigbee_data)
    print(pt_data)
    print(pt_data1)
    print(mic_check)
    # scapy_packet = Dot15d4(data)
    # result = kbdecrypt(scapy_packet, key)
    # print(result)

zigbee_aps_decrypt(data)

`

`
#result
python test.py
35
b"\xd0\xa5`\x14l\x9f\xaa\xf6\xd4\xc6\x86\xf1)\xd1\xc3\x85 W\xf4j\xc1i\xe7\xee\xe8+\xfbV'$ZO\xf4\xf1N"
b"\xd0\xa5`\x14l\x9f\xaa\xf6\xd4\xc6\x86\xf1)\xd1\xc3\x85 W\xf4j\xc1i\xe7\xee\xe8+\xfbV'$ZO\xf4\xf1N\x7f\xf3,\xc1\xfb\xe0\xaa\xd3\xbe\xb9J\x13K"
0

`

taylorcenters, 2022-05-20:
[image]

I saved the packet you provided as a pcap to view in wireshark. It also noted an issue with the source address and did not seem to be able to parse or decrypt the packet

```
>>> bytes(packet)
b'a\x88\xec!8\xb8o\x00\x00\x08\x00\xb8o\x00\x00\x1e~!\xc40\t\xa0\x05\x00\x94J\x84\xfe\xff\xbd\x1b\xec\xdd\xae/>\xde\x19Ff\xb5\xdeHuH\xec\x98B] \x12\xfa\xb4]8>Y\xc87\xa9y0\xc6p\xce\x16\xefM\xdf.\xb97\xeb'
```

jbgod, 2022-05-20:
i captue it from ubiqua,the transport key packet with seqnum 0xec
i had save it as pcap file.
[result.zip](https://github.com/riverloopsec/killerbee/files/8742020/result.zip)
[image]

taylorcenters, 2022-05-20:
Thanks I'll take a look!

jbgod, 2022-05-20:
ok,thank you,i work for many days for this thing,i will keep on this weekend.
i am waiting for your response

jbgod, 2022-05-21:
how to write a Api for Killerbee to call the function zbee_sec_hash in SecBee,
i think the trust center link key should be Hashed by MMO before encrypt the transport message
[https://github.com/Cognosec/SecBee/blob/master/ZigBee%20Crypto/zigbee_transkey.c](url)

taylorcenters, 2022-05-23:
actually, I think I got this working with the hashed key. 

```
>>> from zigbee_crypt import *
>>> from killerbee.scapy_extensions import *
>>>
>>> key = b'ZigBeeAlliance09'
>>> hkey = sec_key_hash(key, b'\x00')
>>>
>>> pkt_list = kbrdpcap('result.pcap')
>>> pkt = pkt_list[131].copy()
>>> bytes(pkt)
b'a\x88\xe9!8\xb8o\x00\x00\x08\x00\xb8o\x00\x00\x1e|!\xc30\x08\xa0\x05\x00\x94J\x84\xfe\xff\xbd\x1b\xec\x19\xee\xc7A\xa3Y*\xe81\xcf\xa0\xe6\xf2\x14\xea\xb9\xae\x81\x07\x100\xc7\x8e\x82\xc4g\xbax\x9fD\xae4\x07w\xb6\x9b?xo\xff\xff'
>>>
>>> pkt.nwk_seclevel = 5
>>>
>>> nonce = struct.pack('L', pkt[ZigbeeSecurityHeader].source)
>>> nonce = nonce + struct.pack('I', pkt[ZigbeeSecurityHeader].fc)
>>> nonce = nonce + bytes(pkt[ZigbeeSecurityHeader])[0:1]
>>>
>>> a_data = bytes(pkt[ZigbeeAppDataPayload])[:-len(pkt.data)]
>>> mic = pkt.data[-4:]
>>> ct_data = pkt.data[:-4]
>>>
>>> (pt_data, check) = decrypt_ccm(hkey, nonce, mic, ct_data, a_data)
>>> pt_data
b'\x05\x01G\xf3 \x01\x83\x1c\x1c\xb6C\xa1E\x7f?\x80\xd9\x9d\x00\x881\xca\xfe\xff\xf9\xe3\xb4\x94J\x84\xfe\xff\xbd\x1b\xec'
>>> check
1
```

I need to put in a PR to fix the issue with missing ext_src to pull source addr from the ZigbeeSecurityHeader field
but hopefully the above will work for you.

jbgod, 2022-05-24:
yes,i solved the aps_payload decrypt.
the secutiry level in the packet is lv 0,but Mic is 4 bytes,so we need to use lv1 or lv 5 to decrypt
[image]
i think this is the little trick of the zigbee device manufacturer

my decrypt code like this
```
from Crypto.Cipher import AES
from zigbee_crypt import *
from killerbee.scapy_extensions import *

def byte_xor(data1, data2):
    result = b''
    for i in range(len(data1)):
        result += bytes([data1[i] ^ data2[i]])
    return result

def pad(x):
    n=(16-len(x)%16)%16
    return x + bytes([0x00]*n)

def zigbee_aps_decrypt(data):   
    mac_payload = data[9 : -2]
    nwk_header = mac_payload[:8]
    nwk_payload = mac_payload[8:]
    aps_header = nwk_payload[:2]
    aps_aux_header = nwk_payload[2:15]
    aps_security_control = aps_aux_header[0]
    aps_frame_counter = aps_aux_header[1:5]
    source_address = aps_aux_header[5:]
    aps_payload = nwk_payload[15:-4]
    aps_mic = nwk_payload[-4:]
    #nonce = source_address + aps_frame_counter + b'\x30'
    #nonce
    #packet send APS security level 0,but the security level is 5,
    #so aps security control == 0x35
    nonce = source_address + aps_frame_counter + b'\x35'

    Flags = bytes ([0b00000001])
    key = bytes([0x5A, 0x69, 0x67, 0x42, 0x65, 0x65, 0x41, 0x6C, 0x6C,0x69,0x61,0x6E,0x63, 0x65, 0x30, 0x39]) 
    pad_enc_data = pad(aps_payload)
    hash_key = sec_key_hash(key, b'\x00')
    aes = AES.new(hash_key)
    #aes = AES.new(key)
    dec_data = b''

    for i in range(len(pad_enc_data) // 16):
        #Ai = Flags + nonce + b'\x00' + b'\x0i' from A1 -> An
        tmp_a = Flags + nonce + bytes([0x00]) + bytes([i + 1])
        mask = aes.encrypt(tmp_a)
        e_data = pad_enc_data[16 * i : 16 * (i + 1)]
        result = byte_xor(mask, e_data)
        dec_data += result

    dec_data = dec_data[:len(aps_payload)]
    return dec_data

enc_data = bytes.fromhex('6188EC2138B86F00000800B86F00001E7E21C43009A00500944A84FEFFBD1BECDDAE2F3EDE194666B5DE487548EC98425D2012FAB45D383E59C837A97930C670CE16EF4DDF2EB9FFFF')
real_data = bytes.fromhex("050147F32001831C1CB643A1457F3F80D99D008831CAFEFFF9E3B4944A84FEFFBD1BEC")
dec_data = zigbee_aps_decrypt(enc_data)
print("decrypt data is")
print(dec_data)
print("real data is")
print(real_data)
```python

jbgod, 2022-05-24:
i just have one question can't solved,the first question of this issue. why the association request i send by killerbee is be reject by the  zigbee gateway
https://github.com/riverloopsec/killerbee/issues/256#issue-1241311966

taylorcenters, 2022-05-24:
What is mac_payload? can you show an example of what payload is that you send to kb_kb.inject(payload) ?

jbgod, 2022-05-25:
[image]
after reveive the beacon back from gateway. end device send a association request to the gateway.
In the association request packet,the Mac Payload is b'\x01\x8e'

[image]
after send a association request, end device will send a data request, the Mac Payload is b'\x04'
attention, this type of target will cause error in  Dot15d4,packet's data don't have source pan id

after send association request and data request to the zigbee coordinator
the coordinator will response a association response，it show the status of joining a pan.
[image]
this is association response packlet with success 

https://user-images.githubusercontent.com/20832503/169230205-b9d03077-fae8-4d99-901d-759f21aefaa6.png
this is assocaiton response packet with deny

closed 2022-05-27 by jbgod