riverloopsec/killerbee #258: PY_SSIZE_T_CLEAN required with Python 3.10

doegox, 2022-06-02:
Hi

Since a while, a warning was emitted about defining PY_SSIZE_T_CLEAN but now with Python 3.10 it's a breaking error:
```
  File ".../venv/lib/python3.10/site-packages/killerbee-3.0.0b2-py3.10-linux-x86_64.egg/killerbee/scapy_extensions.py", line 482, in kbdecrypt
    (payload, micCheck) = zigbee_crypt.decrypt_ccm(key, nonce, pkt.mic, encrypted, zigbeeData)
SystemError: PY_SSIZE_T_CLEAN macro must be defined for '#' formats
```

I was lazy to fork just for a MR, but here is a bugfix in attachment.
[fixtype.diff.txt](https://github.com/riverloopsec/killerbee/files/8827494/fixtype.diff.txt)

JuliaKulacz, 2022-10-17:
Hi,
Did you mange to resolve this issue?

doegox, 2022-10-17:
Well when I apply locally the bugfix I suggested to you, yes it works. So please apply my proposed bugfix.
```diff
diff --git a/zigbee_crypt/zigbee_crypt.c b/zigbee_crypt/zigbee_crypt.c
index 4b3034b..8bf9e9c 100644
--- a/zigbee_crypt/zigbee_crypt.c
+++ b/zigbee_crypt/zigbee_crypt.c
@@ -10,6 +10,7 @@
 
 // Explaination of Python Build Values http://docs.python.org/c-api/arg.html#Py_BuildValue
 
+#define PY_SSIZE_T_CLEAN
 #include <Python.h>
 #include <stdio.h>
 #include <gcrypt.h>
@@ -43,14 +44,14 @@
 static PyObject *zigbee_crypt_encrypt_ccm(PyObject *self, PyObject *args) {
 	// This was modeled after zigbee_crypt_decrypt_ccm in reverse
 	const char			*pZkey;
-	int					sizeZkey;
+	Py_ssize_t			sizeZkey;
 	const char			*pNonce;
-	int					sizeNonce;
-	int					sizeMIC;
+	Py_ssize_t			sizeNonce;
+	Py_ssize_t			sizeMIC;
 	const char			*pUnencryptedData;
-	int					sizeUnencryptedData;
+	Py_ssize_t			sizeUnencryptedData;
 	const char			*zigbeeData;
-	int					sizeZigbeeData;
+	Py_ssize_t			sizeZigbeeData;
 	int i, j;
 	PyObject			*res;
 
@@ -63,9 +64,9 @@ static PyObject *zigbee_crypt_encrypt_ccm(PyObject *self, PyObject *args) {
 	gcry_cipher_hd_t	cipher_hd;
 
 #if PY_MAJOR_VERSION >= 3
-	if (!PyArg_ParseTuple(args, "y#y#iy#y#",
+	if (!PyArg_ParseTuple(args, "y#y#ny#y#",
 #else
-	if (!PyArg_ParseTuple(args, "s#s#is#s#",
+	if (!PyArg_ParseTuple(args, "s#s#ns#s#",
 #endif
 								&pZkey, &sizeZkey,
 								&pNonce, &sizeNonce,
@@ -247,15 +248,15 @@ static PyObject *zigbee_crypt_encrypt_ccm(PyObject *self, PyObject *args) {
 
 static PyObject *zigbee_crypt_decrypt_ccm(PyObject *self, PyObject *args) {
 	const char			*pZkey;
-	int					sizeZkey;
+	Py_ssize_t			sizeZkey;
 	const char			*pNonce;
-	int					sizeNonce;
+	Py_ssize_t			sizeNonce;
 	const char			*pOldMIC;
-	int					sizeMIC;
+	Py_ssize_t			sizeMIC;
 	const char			*pEncryptedData;
-	int					sizeEncryptedData;
+	Py_ssize_t			sizeEncryptedData;
 	const char			*zigbeeData;
-	int					sizeZigbeeData;
+	Py_ssize_t			sizeZigbeeData;
 	PyObject			*res;
 
 	char				pMIC[ZBEE_SEC_CONST_MICSIZE];
```