riverloopsec/killerbee #262: [Help required]

jbasement, 2022-09-03:
Hello everyone,

this is not a technical issue but more a request for support/ help.
I was looking through the supported devices for KillerBee that also allow sending packages, like the ApiMote, Tmote sky or the Atmel RZ Raven USBSTICK. But none of those are available to purchase.
As I'm currently starting to write my bachelor thesis on this topic and I have to do some tests with KillerBee I wanted to ask for help where I could buy a support KillerBee device?
I'm just a regular student so I wont be able to build an ApiMote myself.

Thanks in adavance

jbasement, 2022-09-03:
Update:
I just found out, by sifting through old issues, that CC2531 devices are supported when using [BumbleBee firmware by virtualabs](https://github.com/virtualabs/cc2531-killerbee-fw). This wasn't listed on the firmware/readme.md. 
I can create a draft PR to update that readme but I think I'm missing the expertise to properly add the info there.

declanCoster, 2022-09-05:
https://www.attify-store.com/products/apimote

jbasement, 2022-09-05:
@declanCoster thanks for your reply. But at least for me it shows up as "Sold out".

declanCoster, 2022-09-05:
Oh shoot, sorry didn't see that

jbasement, 2022-09-27:
Created a PR for that https://github.com/riverloopsec/killerbee/pull/266

closed 2022-12-21 by jbasement