riverloopsec/killerbee #265: APIMOTE and RZRaven USB

ACIS-Chindanai, 2022-09-24:
I am currently learning about the Zigbee protocol.
I try to find where I can buy APIMOTE and RZRaven USB but every shop is sold out.
              
Does anyone have a apimote or rzraven usb?
Would you mind if you sent it to me?
A brand new one or demo or pre-owned, no problem.
I can pay for it and the shipping cost.(Thailand)

jbasement, 2022-09-26:
Hi @ACIS-Chindanai 
I was just facing the same problem a few weeks back but managed to find a solution. 
The CC2531 (~10-15€) is still available online and is compatible with KillerBee. With the default firmware it only supports sniffing. But @virtualabs was so gracious to develop a [driver for CC2531](https://github.com/virtualabs/cc2531-killerbee-fw) that allows sending/ injecting packets as well.
Hope that helps!

ACIS-Chindanai, 2022-09-27:
Hi @jbasement 
Regards, bro. I've had a CC2531 for a long, and I'm using the Bumblebee 2.1r1 firmware, but I'm unable to sniff with zbwireshark or zbstumbler because the packet doesn't show up.
However, ZBOSS sniffer worked for me.

The only problem is how to inject and attack the zigbee. Because I can't use command zbwireshark so I don't have Draintree file to zbreplay.

Please guide me.
Thank you

virtualabs, 2022-09-27:
If you think Bumblebee doesn't work as expected or at least has an issue, please file an issue in the project repository. For the record, Bumblebee cannot sniff packets with bad FCS (if I remember correctly), so it is a possible explanation of why your packet did not show up (but it could also because of Bumblebee as well).