riverloopsec/killerbee #269: Zbjammer on cc2531

remydemy, 2022-11-18:
Hello,

I'm trying to use the zbjammer with the cc2531. I've flashed the cc2531 with the [Bumblebee version 2.1 revision 1](https://github.com/virtualabs/cc2531-killerbee-fw/releases/tag/release-2.1r1) and am using the develop branch of killerbee.

When I'm executing zbjammer I get the following error message:
``` 
❯ zbjammer -c 15
zbjammer: jamming channel 15
*** WARNING: this may not actually work on your hardware! Check with spectrum analyser!
*** NOTICE: it is your responsibility to comply with local law. Please check radio spectrum laws in your area before
    proceeding. Hit <ENTER> to continue or CTL-C to abort.
Traceback (most recent call last):
  File "/Users/emily/.pyenv/versions/3.10.4/bin/zbjammer", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbjammer')
  File "/Users/emily/.pyenv/versions/3.10.4/lib/python3.10/site-packages/pkg_resources/__init__.py", line 651, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/Users/emily/.pyenv/versions/3.10.4/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1448, in run_script
    exec(code, namespace, namespace)
  File "/Users/emily/.pyenv/versions/3.10.4/lib/python3.10/site-packages/killerbee-3.0.0b2-py3.10-macosx-12.2-x86_64.egg/EGG-INFO/scripts/zbjammer", line 51, in <module>
    if not kb.jammer_on():
  File "/Users/emily/.pyenv/versions/3.10.4/lib/python3.10/site-packages/killerbee-3.0.0b2-py3.10-macosx-12.2-x86_64.egg/killerbee/__init__.py", line 459, in jammer_on
    return self.driver.jammer_on(channel=channel, method=method)
TypeError: Bumblebee.jammer_on() got an unexpected keyword argument 'method'
```

Does anybody know what I can do to resolve this issue?

Thanks,
Emily

ArcticNarwhal, 2022-11-23:
Hi, that function hasn't been implemented yet (see [/killerbee/dev_bumblebee.py](https://github.com/riverloopsec/killerbee/blob/748740d1198d9edba55d25614a84e39cf20bf414/killerbee/dev_bumblebee.py))

remydemy, 2022-11-23:
@ArcticNarwhal Yes, I saw that too after I posted this issue, but even when I'm trying to execute the zbdump an error occurs. Well not necessarily an error, but the pcap file is empty, even though the stick lights up when it seems to receive a packet.

I'm trying to capture OpenThread messages. Could that be the problem? I thought it should work because OpenThread also uses IEEE 802.15.4.

I tried to investigate the problem a bit and found that in the `dev_bumblebee.py`, in the variable `rx_buffer` bytes are being stored (I assume those are the packets), but in the method `process_packet()`, where the packets are read there seems to be some kind of problem.

Do you know what that could be?

remydemy, 2022-11-23:
Ok nvm if I use version 2.0 of bumblebee and the current version of killerbee zbdump works just fine!