riverloopsec/killerbee #273: Pycrypto - deprecated

DuckSounds, 2023-04-17:
Finally installed Killerbee on my Kali due to enormous amount of error regarding newer/older versions. 
Is there any fix for 'pycrypto' as it's not a suitable thing to be used anymore?

I installed python3-dev and pycryptodome instead as per some guides but none of the commands can be run due to it expecting pycrypto.

zbid        
Traceback (most recent call last):
  File "/usr/local/bin/zbid", line 4, in <module>
    __import__('pkg_resources').run_script('killerbee==3.0.0b2', 'zbid')
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
     ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 3264, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'pycrypto' distribution was not found and is required by killerbee

DuckSounds, 2023-04-17:
Solved this by changing setup.py file to use pycryptodome instead of pycrypto, will see if this works but it's not returning error above.

tc-26, 2023-05-31:
Thank you for reporting, we should definitely move off of unsupported dependencies. I'll try with pycryptodome as well and see if it's that easy of a switch or if there are issues laying in wait

RandomSignals, 2023-07-04:
Any updates on this moving to supported versions?

jamesleytevidal, 2024-10-01:
I used pycryptodome the last time I installed this on a new machine and it's worked fine so far, but I haven't tested all the utilities exhaustively.